panoptes/POCS #420: Disentangle reading from multiple arduinos

jamessynge, 2018-01-27:
ArduinoSerialMonitor supports reading from both the telemetry and camera boards, but effectively requires them to produce output at exactly the same rate else a data loss will occur.

I'm planning to create pocs/sensors/arduino_io.py, based on peas/sensors.py, but the primary class will support a single serial connected device.

jamessynge, 2018-01-27:
Related to #326, we should consider how to find the available relays on each board. One option would be for the json output to tell us. For example:
`... "relays": {"camera": 1, "fan": 0, ...}, ...`
Then we can send to the board a name that it produced:
`camera=0\n`

jamessynge, 2018-02-06:
I've created ArduinoIO and SensorRecorder (PR #428), designed for working in a multithreaded Python app. However, I've just learned that debugging such an app isn't possible using Python's native debugger, pdb. While I could debug it using log messages, my inclination is to instead head in a direction that Wilfred mentioned: use separate processes for each Arduino. Specifically:

- Each process works with a specific device (e.g. `/dev/ttyACM1`).
- PanDB is used to save each reading into a history collection and a current collection, named with the name of the board.
- PanMessaging is used for two purposes:
  - Sending readings to the rest of the system on channel environment.
  - Receiving relay change commands (format to be determined, though probably using a channel name that includes the name of the board).
- The process stops if the device stops working; this allows a supervising process to re-do auto-detection periodically and find the board attached to a new device. While it would be useful for a message to be sent indicating that there is an issue, zeromq doesn't guarantee delivery, so we might want to rely on other means for detecting that the connection has been lost.

wtgee, 2018-02-07:
Think this sounds like a good move and would also be easy to implement for the Pi.

jamessynge, 2018-02-11:
PR coming in a few minutes. The ArduinoIO object listens for commands to channel `<board>:commands` (e.g. `telemetry_board:commands`), with the following incoming messages:

`{"command": "shutdown"}` - Stops the main loop.

`{"command": "write_line", "line": "some line of text"}` - Sends the line of text to the Arduino (serial port)

I plan to extend this (once we agree on the protocol) to support changing relays by name (already supported by the telemetry board sketch). That makes it easier to replace the Arduino with a Raspberry Pi or other maker board. That might mean that we should change the name from ArduinoIO to SerialSensorIO, etc.

wtgee, 2018-02-13:
Linking #125

jamessynge, 2018-09-15:
Resuming work on this, with this plan:

- [ ] Add DelaySigTerm, a context manager for preventing SIGTERM from interrupting a critical section. If SIGTERM is caught, it is re-sent after the critical section.
- [ ] Use DelaySigTerm in ArduinoIO so that the state isn't corrupted by SIGTERM, which is used to stop a process running ArduinoIO.
- [ ] Update arduino_recorder.py script (runs ArduinoIO locally) to support finding messaging ports from config file.
- [ ] Add SensorManager, a process for managing subprocesses running arduino_recorder.py, one for each configured Arduino board.
- [ ] Integrate SensorManager into the running of pocs_shell and/or peas_shell. This item will need work to clarify. Ideally, we'd introduce something inspired by supervisord.

wtgee, 2020-02-17:
Mostly closed by #880. Could use some cleanup, but closing this issue as stale.

closed 2020-02-17 by wtgee