panoptes/POCS #431: Reduce use of, or requirement for, environment variables

jamessynge, 2018-01-31:
As a unit maintainer, I don't want to have to update `$HOME/.profile` each time we change the set of env vars. As a developer, I'd like to be able to run pytest from the root directory of POCS or other repos, without having to run `python setup.py develop`. This enables using git worktrees (i.e. different directory trees for different branches), and thus makes it easier to work on two or more features at once or to check that the develop branch is working, as I did last night... it wasn't.

There are currently 6 env vars in scripts/install/default-env-vars.sh. We could certainly have many of these env vars found by discovery (assuming not set).

Furthermore, if we're running from POCS, matplotlib will look for file matplotlibrc in the current directory. By instead setting MATPLOTLIBRC (as in PR #430), we are making the POCS file apply to all applications that the user runs. This will be confusing for folks trying to run some other program that uses matplotlib.

wtgee, 2018-02-01:
We could certainly reduce some although the point of the separate file was that you could `source` from your `.profile` once and then not need to change it (although see below). I also don't foresee them changing much and indeed haven't actually changed in 3+ years so I'm not sure what the concern is. The 5 panoptes env vars are also actually supporting three applications (POCS, PIAA, PAWS) but are set up in one file for convenience.

Running from root of a repo to support git worktrees is a separate issue and if important should be addressed as a separate issue, as is a develop branch not working (not sure what that actually means).

MATPLOTLIBRC is added as a convenience. If you have set your matplotlibrc up for you development environment then it's not needed. The `default-env-vars.sh` is not currently run from anywhere precisely because of this concern, as was pointed out in the original PR #426.

jamessynge, 2018-02-01:
I use the following in my `.profile` file:

```bash
# Set the Panoptes variables to their default values if not set.
[[ -z "$PANUSER" ]] && export PANUSER=panoptes     # NOT TRUE ON MY LAPTOP
[[ -z "$PANDIR" ]] && export PANDIR=/var/panoptes  # Main Dir
[[ -z "$PANLOG" ]] && export PANLOG=$PANDIR/logs   # Log files
[[ -z "$POCS" ]] && export POCS=$PANDIR/POCS       # Main Observatory Control
[[ -z "$PAWS" ]] && export PAWS=$PANDIR/PAWS       # Web Interface
[[ -z "$MATPLOTLIBRC" ]] && export MATPLOTLIBRC=${POCS}/.matplotlib
```

This allows for only setting the env var if it has not already been set. Perhaps we can do something similar in `default-env-vars.sh`?

wtgee, 2018-02-01:
I don't understand why that would be needed and haven't ever seen that before in a profile or rc file. Why would you ever need the check? I'm not opposed to it just don't see where it would ever come into play versus just setting it. I guess for the matplotlib if a user has already set that variable earlier?

wtgee, 2018-02-13:
I think we've cleaned this up or resolved it somewhat. @jamessynge ok to close?

closed 2018-02-20 by wtgee