panoptes/POCS #435: Enable pytest to run in parallel

jamessynge, 2018-02-03:
The pytest-xdist plugin supports running tests in multiple processes, but it appears that our own logging code causes those tests to fail. I had much better luck adding --boxed to the command line below, but sometimes still see failures like those below.

```python
$ pytest -n=auto --timeout=90
======================================================================= test session starts =======================================================================
platform linux -- Python 3.6.3, pytest-3.2.5, py-1.5.2, pluggy-0.4.0
rootdir: /home/james/git/panoptes/POCS, inifile:
plugins: xdist-1.20.1, timeout-1.2.1, forked-0.2, cov-2.5.1
timeout: 90.0s method: signal
gw0 [371] / gw1 [371] / gw2 [369] / gw3 [369] / gw4 [369] / gw5 [371] / gw6 [369] / gw7 [371]                                                                      
scheduling tests via LoadScheduling
collecting 0 items / 8 errors                                                                                                                                      
============================================================================= ERRORS ==============================================================================
___________________________________________________________ ERROR collecting peas/tests/test_boards.py ____________________________________________________________
peas/tests/test_boards.py:3: in <module>
    from peas.sensors import ArduinoSerialMonitor
peas/sensors.py:14: in <module>
    from pocs.utils.messaging import PanMessaging
pocs/utils/messaging.py:16: in <module>
    class PanMessaging(object):
pocs/utils/messaging.py:22: in PanMessaging
    logger = get_root_logger()
pocs/utils/logger.py:108: in get_root_logger
    os.symlink(log_symlink_target, log_symlink)
E   FileExistsError: [Errno 17] File exists: 'per-run/-c/-c-20180203T203136Z-24427-warn.log' -> '/var/panoptes/logs/-c-warn.log'
___________________________________________________________ ERROR collecting peas/tests/test_boards.py ____________________________________________________________
peas/tests/test_boards.py:3: in <module>
    from peas.sensors import ArduinoSerialMonitor
peas/sensors.py:14: in <module>
    from pocs.utils.messaging import PanMessaging
pocs/utils/messaging.py:16: in <module>
    class PanMessaging(object):
pocs/utils/messaging.py:22: in PanMessaging
    logger = get_root_logger()
pocs/utils/logger.py:108: in get_root_logger
    os.symlink(log_symlink_target, log_symlink)
E   FileExistsError: [Errno 17] File exists: 'per-run/-c/-c-20180203T203136Z-24423-all.log' -> '/var/panoptes/logs/-c-all.log'
___________________________________________________________ ERROR collecting peas/tests/test_boards.py ____________________________________________________________
peas/tests/test_boards.py:3: in <module>
    from peas.sensors import ArduinoSerialMonitor
peas/sensors.py:14: in <module>
    from pocs.utils.messaging import PanMessaging
pocs/utils/messaging.py:16: in <module>
    class PanMessaging(object):
pocs/utils/messaging.py:22: in PanMessaging
    logger = get_root_logger()
pocs/utils/logger.py:108: in get_root_logger
    os.symlink(log_symlink_target, log_symlink)
E   FileExistsError: [Errno 17] File exists: 'per-run/-c/-c-20180203T203136Z-24435-all.log' -> '/var/panoptes/logs/-c-all.log'
___________________________________________________________ ERROR collecting peas/tests/test_boards.py ____________________________________________________________
peas/tests/test_boards.py:3: in <module>
    from peas.sensors import ArduinoSerialMonitor
peas/sensors.py:14: in <module>
    from pocs.utils.messaging import PanMessaging
pocs/utils/messaging.py:16: in <module>
    class PanMessaging(object):
pocs/utils/messaging.py:22: in PanMessaging
    logger = get_root_logger()
pocs/utils/logger.py:108: in get_root_logger
    os.symlink(log_symlink_target, log_symlink)
E   FileExistsError: [Errno 17] File exists: 'per-run/-c/-c-20180203T203136Z-24419-all.log' -> '/var/panoptes/logs/-c-all.log'
______________________________________________________________________ ERROR collecting gw0 _______________________________________________________________________
Different tests were collected between gw2 and gw0. The difference is:
--- gw2

+++ gw0

@@ -1,3 +1,5 @@

+peas/tests/test_boards.py::test_create
+peas/tests/test_boards.py::test_has_readers
 peas/tests/test_sensors.py::test_create_camera_simulator
 peas/tests/test_sensors.py::test_create_default_simulator
 peas/tests/test_sensors.py::test_detect_board_on_port_not_a_board
______________________________________________________________________ ERROR collecting gw1 _______________________________________________________________________
Different tests were collected between gw2 and gw1. The difference is:
--- gw2

+++ gw1

@@ -1,3 +1,5 @@

+peas/tests/test_boards.py::test_create
+peas/tests/test_boards.py::test_has_readers
 peas/tests/test_sensors.py::test_create_camera_simulator
 peas/tests/test_sensors.py::test_create_default_simulator
 peas/tests/test_sensors.py::test_detect_board_on_port_not_a_board
______________________________________________________________________ ERROR collecting gw7 _______________________________________________________________________
Different tests were collected between gw2 and gw7. The difference is:
--- gw2

+++ gw7

@@ -1,3 +1,5 @@

+peas/tests/test_boards.py::test_create
+peas/tests/test_boards.py::test_has_readers
 peas/tests/test_sensors.py::test_create_camera_simulator
 peas/tests/test_sensors.py::test_create_default_simulator
 peas/tests/test_sensors.py::test_detect_board_on_port_not_a_board
______________________________________________________________________ ERROR collecting gw5 _______________________________________________________________________
Different tests were collected between gw2 and gw5. The difference is:
--- gw2

+++ gw5

@@ -1,3 +1,5 @@

+peas/tests/test_boards.py::test_create
+peas/tests/test_boards.py::test_has_readers
 peas/tests/test_sensors.py::test_create_camera_simulator
 peas/tests/test_sensors.py::test_create_default_simulator
 peas/tests/test_sensors.py::test_detect_board_on_port_not_a_board
===================================================================== 8 error in 6.71 seconds =====================================================================
```

wtgee, 2018-02-04:
If it's just symlinking the logs maybe we can catch and ignore that? Would be nice to see this used somehow.

wtgee, 2020-02-17:
Related to #520 and #191