panoptes/POCS #437: Tests can break running pocs_shell on a unit

jamessynge, 2018-02-04:
I ran tests on PAN006 this afternoon and later noticed in the log file that POCS had stopped, apparently because of a shutdown message sent by a test (i.e. it appears that the tests and the "real" code use the same ports).

```
D0203 21:02:20.101 core.py:337          Dark Check: False
D0203 21:02:20.102 core.py:373          Weather Safety: True [15 sec old - 2018-02-03 21:02:05.250000]
D0203 21:07:25.113 observatory.py:82    Sun 8.31 deg > -18.0 deg
D0203 21:07:25.113 core.py:337          Dark Check: False
D0203 21:07:25.114 core.py:373          Weather Safety: True [71 sec old - 2018-02-03 21:06:14.119000]
D0203 21:12:30.116 observatory.py:82    Sun 7.49 deg > -18.0 deg
D0203 21:12:30.116 core.py:337          Dark Check: False
D0203 21:12:30.117 core.py:373          Weather Safety: True [44 sec old - 2018-02-03 21:11:45.968000]
D0203 21:17:35.097 observatory.py:82    Sun 6.67 deg > -18.0 deg
D0203 21:17:35.097 core.py:337          Dark Check: False
D0203 21:17:35.098 core.py:373          Weather Safety: True [17 sec old - 2018-02-03 21:17:17.807000]
D0203 21:22:40.154 observatory.py:82    Sun 5.84 deg > -18.0 deg
D0203 21:22:40.155 core.py:337          Dark Check: False
D0203 21:22:40.156 core.py:373          Weather Safety: True [73 sec old - 2018-02-03 21:21:26.700000]
I0203 21:27:04.917 core.py:493          Message received: command shutdown
W0203 21:27:04.917 core.py:510          Shutdown command received
I0203 21:27:04.919 messaging.py:122     PANCHAT I'm powering down
I0203 21:27:04.919 core.py:226          Shutting down POCS State Machine: , please be patient and allow for exit.
D0203 21:27:04.919 abstract_serial_dome.py:70 Already connected to dome
D0203 21:27:06.017 astrohaven.py:183    Got target_feedback, feedback_countdown=1
D0203 21:27:06.426 astrohaven.py:183    Got target_feedback, feedback_countdown=0
D0203 21:27:08.339 astrohaven.py:183    Got target_feedback, feedback_countdown=1
D0203 21:27:08.748 astrohaven.py:183    Got target_feedback, feedback_countdown=0
D0203 21:27:11.647 observatory.py:125   Shutting down observatory
D0203 21:27:11.647 serial.py:64         Closing serial port for mount
D0203 21:27:11.648 rs232.py:151         SerialData.disconnect called for /dev/ttyUSB0
D0203 21:27:11.648 abstract_serial_dome.py:75 Closing serial port for dome
D0203 21:27:11.648 rs232.py:151         SerialData.disconnect called for /dev/ttyUSB2
D0203 21:27:11.648 core.py:254          Shutting down messaging system
D0203 21:27:11.648 core.py:258          Terminating check_messages - PID 22606
D0203 21:27:11.652 core.py:258          Terminating cmd_forwarder - PID 22594
I0203 21:27:11.671 core.py:264          Power down complete
D0203 21:27:11.672 mount.py:116         Problem getting mount status: 
W0203 21:27:11.673 observatory.py:171   Can't get observatory status: 
I0203 21:27:11.674 core.py:493          Message received: command shutdown
W0203 21:27:11.675 core.py:510          Shutdown command received
D0203 21:27:11.676 mount.py:116         Problem getting mount status: 
W0203 21:27:11.676 observatory.py:171   Can't get observatory status: 
D0203 21:27:11.678 mount.py:116         Problem getting mount status: 
W0203 21:27:11.678 observatory.py:171   Can't get observatory status: 
D0203 21:27:11.683 mount.py:116         Problem getting mount status: 
W0203 21:27:11.683 observatory.py:171   Can't get observatory status: 
D0203 21:27:11.685 mount.py:116         Problem getting mount status: 
W0203 21:27:11.686 observatory.py:171   Can't get observatory status: 
D0203 21:27:11.693 mount.py:116         Problem getting mount status: 
W0203 21:27:11.693 observatory.py:171   Can't get observatory status: 
```

wtgee, 2018-02-04:
Yikes. Makes perfect sense but obviously shouldn't be that way. This and #131 are some arguments for getting that messaging cleaned up sooner than later.

wtgee, 2018-02-19:
Closed as part of #440

closed 2018-02-19 by wtgee