panoptes/POCS #446: PanDB search date range

wtgee, 2018-02-05:
Need to add the ability to search a date range, mostly for PanFileDB. See for example #442

jamessynge, 2018-02-10:
I was thinking something similar as I was reviewing the database code and uses of PanDB throughout POCS. I noticed that there is some access into the guts of PanMongoDB to read from the collections.

wtgee, 2018-02-10:
>  I noticed that there is some access into the guts of PanMongoDB to read from the collections.
 
You mean outside of the PanDB abstraction? I added a quick fix in the `plot_weather.py` script the other day but thought I had removed all the rest of them.

wtgee, 2020-02-17:
We don't do any date search on the local data and it's not clear it would be needed. Closing as stale.

closed 2020-02-17 by wtgee