panoptes/POCS #450: catch camera timeout

lspitler, 2018-02-06:
For SBIGs it seems to happen on readout.

wtgee, 2018-02-08:
In general we perhaps need something in the `camera.process_exposure()` method that times out events after a while (probably set up in the `take_exposure` methods).

Note that for Huntsman specifically we have an omission in the `observing` state where we are incrementing the timer but never acting on it. See huntsman observing [here](https://github.com/AstroHuntsman/huntsman-pocs/blob/develop/states/huntsman/observing.py#L23) compared to the breakout in the panoptes observing [here](https://github.com/panoptes/POCS/blob/develop/pocs/state/states/default/observing.py#L28).

wtgee, 2018-02-08:
Also just a note that we should obviously fix the Huntsman observing state but we experience them most often in the autofocus routines.

AnthonyHorton, 2018-02-08:
Some attempts to handle timeouts more gracefully in https://github.com/panoptes/POCS/pull/458

AnthonyHorton, 2018-02-12:
Could use some fresh eyes on this.  Based on the logs it looks like execution is getting stuck in this section of code in `pocs/camera/sbigudrv.py`:

```
        # Readout data
        with self._command_lock:
            self._set_handle(handle)
            self._send_command('CC_END_EXPOSURE', params=end_exposure_params)
            self._send_command('CC_START_READOUT', params=start_readout_params)
            for i in range(height):
                try:
                    self._send_command('CC_READOUT_LINE',
                                       params=readout_line_params,
                                       results=as_ctypes(image_data[i]))
                except RuntimeError as err:
                    message = 'Readout error on {}: expected {} rows, got{}!'.format(self._ccd_info[handle]['serial number'],
                                                                                     height,
                                                                                     i)
                    self.logger.error(message)
                    self.logger.error(err)
                    warn(message)
                    break

            try:
                self._send_command('CC_END_READOUT', params=end_readout_params)
            except RuntimeError as err:
                message = "Error ending readout on {}: {}".format(self._ccd_info[handle]['serial number'],
                                                                  err)
                self.logger.error(message)
            else:
                self.logger.debug('Readout on {} complete'.format(self._ccd_info[handle]['serial number']))
            finally:
                fits_utils.write_fits(image_data, header, filename, self.logger, exposure_event)
```

The driver reports a `CE_RX_TIMEOUT` error from one of the `CC_READOUT_LINE` commands, which causes my code to raise `RuntimeError`, which is caught by the `except` clause, which writes the error details to the log.  That much works.

Execution should then `break` out of the enclosing `for` loop, and `try` to send the `CC_END_READOUT` command.  A message about whether that succeeds or not should then be written to the log, and either way `fits_utils.write_fits()` is called to write the (partial) image to file and, more importantly, set the `exposure_event` which allows execution of the main thread to continue.  This doesn't work.  Neither of the messages about success/failure of the `CC_END_READOUT` command are making it into the log, and the `exposure_event` isn't getting set, but I can't understand how that could be.

Any thoughts, @wtgee? Or anyone else?

wtgee, 2018-02-12:
Will you post the last log message with number of rows. As I recall it was right at the very end, right?

wtgee, 2018-02-12:
We should be able to test this remotely even if not on sky, correct?

AnthonyHorton, 2018-02-12:
Yes. Can just run autofocus a bunch of times. Doesn't matter if the cameras can't see any stars when testing this.

wtgee, 2018-04-03:
@AnthonyHorton will this be mitigated by using a RaspPi? Do we need to worry about the Issue otherwise?

AnthonyHorton, 2018-04-09:
@wtgee The confusion discussed in my comment above from 12th February is still bugging me but, from a purely practical perspective, this is probably all moot now.

The evidence that we have continues to support the theory that the root cause of the SBIG camera timeout issues was simply USB issues (again!), caused by running multiple cameras on one computer. Running the cameras with a RaspPi each will probably avoid the problem.

It would still be good to make the camera stuff more robust with respect to errors in general, but if the RaspPi solution make the problem go away for now then it's no longer a pressing issue.

wtgee, 2018-10-16:
Not really solved but non longer relevant. Closing.

closed 2018-10-16 by wtgee