panoptes/POCS #452: peas_shell exceptions

wtgee, 2018-02-07:
I'm seeing this every second on peas_shell, not sure cause. Will look into and report:

```
W0207 03:24:38.687 sensors.py:119       Exception while reading from sensor camera_board: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
D0207 03:24:38.687 sensors.py:106       ArduinoSerialMonitor.capture reading sensor telemetry_board
D0207 03:24:39.686 sensors.py:112       Got sensor_value from telemetry_board
D0207 03:24:40.691 sensors.py:106       ArduinoSerialMonitor.capture reading sensor camera_board
W0207 03:24:40.691 sensors.py:119       Exception while reading from sensor camera_board: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
D0207 03:24:40.692 sensors.py:106       ArduinoSerialMonitor.capture reading sensor telemetry_board
D0207 03:24:41.687 sensors.py:112       Got sensor_value from telemetry_board
D0207 03:24:42.691 sensors.py:106       ArduinoSerialMonitor.capture reading sensor camera_board

```

wtgee, 2018-02-07:
I've actually lost access to the arduino device in the camera box. The cameras are still appearing but the arduino is uncommunicative even after power cycling and computer reboot. Not sure what the problem is but feel like I have seen it before.

The error appeared in the log, but perhaps the fix would be to have peas_shell somehow communicate that it has lost contact with one of the devices.

jamessynge, 2018-02-10:
Have you been able to resurrect the camera board arduino? If not, can you see it if you run [lsusb]?

wtgee, 2018-02-10:
Lost all access to PAN001 in a storm a few days ago. No one has been able to go up and reset it yet.

jamessynge, 2018-02-12:
FWIW, I've tested the new Arduino IO code when a disconnect occurs. It will disconnect (close) the serial device, then try to reopen. If it is able, and it is the same device, it will keep running. Otherwise, it will die (intentional). My plan is that possible peas_shell will handle this be redoing the auto-detect, finding any attached boards that are not already opened by an Arduino IO.

wtgee, 2018-08-27:
@jamessynge can we close this issue?

wtgee, 2019-06-30:
Closing as stale.

closed 2019-06-30 by wtgee