panoptes/POCS #466: test_dispatch_scheduler.py is flaky

jamessynge, 2018-02-10:
`test_observed_list` has failed sometimes. An example failure:

```
    def test_observed_list(scheduler):
        assert len(scheduler.observed_list) == 0
    
        time = Time('2016-09-11 07:08:00')
        scheduler.get_observation(time=time)
    
        assert len(scheduler.observed_list) == 1
    
        # A few hours later should now be different
        time = Time('2016-09-11 10:30:00')
        scheduler.get_observation(time=time)
    
>       assert len(scheduler.observed_list) == 2
E       AssertionError: assert 1 == 2
E        +  where 1 = len(OrderedDict([('20160909T080000', <pocs.scheduler.observation.Observation object at 0x7f10efd1bf60>)]))
E        +    where OrderedDict([('20160909T080000', <pocs.scheduler.observation.Observation object at 0x7f10efd1bf60>)]) = <pocs.scheduler.dispatch.Scheduler object at 0x7f10efd306a0>.observed_list

pocs/tests/test_dispatch_scheduler.py:237: AssertionError
```

wtgee, 2018-02-10:
I've never seen this error. Could it be related to any work that you are doing?

jamessynge, 2018-02-12:
It's possible. I'll see if I can reproduce on develop.

wtgee, 2018-10-16:
Closing as stale, please reopen if needed.

closed 2018-10-16 by wtgee