panoptes/POCS #474: Decide on `conda install` or `pip install`

jamessynge, 2018-02-11:
There are conda packages (and channels) for a subset of our requirements, but they have the advantage that they are designed to work together. I recently added support for installing pandas via conda, but then discovered that it is also supported by pip. By installing with pip, I ended up with two installs of numpy.

So, we should decide which works better for us, especially if we want to support non-intel computers (e.g. the Raspberry Pi).

Assigning to several folks in an effort to get a discussion going.

jamessynge, 2018-02-11:
FYI, Wilfred tells me that he has experimented with creating a panoptes channel within the conda system, so that unit builders can do an install of many dependencies via conda, including POCS itself.

wtgee, 2018-02-13:
Note that the 3rd option is the `setup.py` install. I would favor that over `pip install` and would also allow us to support features (e.g. #196) [docs here](http://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-extras-optional-features-with-their-own-dependencies)

jamessynge, 2018-09-09:
I've discussed a bit with @wtgee, during which we've pretty much decided that we're going to use conda in preference to pip, where possible. This will enable transition in the future to a conda channel, which exists but isn't populated with the packages we need yet. A nice feature of conda packages is that they can support multiple architectures, so we could have packages that support both the NUC and the RPi.

@wtgee and @AnthonyHorton, any objections to this?

wtgee, 2018-09-10:
I think it is okay. POCS is not very useful as a module on it's own (i.e. will never be on PyPI), it relies on having an environment set up for it, and I think that will be much easier to provide via conda.

wtgee, 2020-02-17:
Will be closed by #951  (or at least made stale).

closed 2020-02-17 by wtgee