panoptes/POCS #477: Allow variations in dark check

jamessynge, 2018-02-12:
For taking "pretty pictures", rather than transit hunting, it would be nice to be able to allow a different dark check. For example, it looks nice and dark to me now, with Orion (a nice target) high in the sky, but it is not yet dark enough for panoptes to wake up.

wtgee, 2018-02-12:
This is specified in the config: https://github.com/panoptes/POCS/blob/develop/conf_files/pocs.yaml#L10

We would ideally take twilight flats from 0° until it was too dark and then maybe fire this up. So there will be some change with the above setting depending on that.

wtgee, 2018-02-19:
Closing unless more specific need than config item is identified.

closed 2018-02-19 by wtgee