panoptes/POCS #478: Camera simulator

wtgee, 2018-02-13:
Currently the camera simulator generates some random data but then changes the filename so that the `solved.fits` is used during processing. This is having some strange side-effects in various places. Additionally, trying to just `run` pocs with the simulators leads to a solve-field error which means it hangs after the `pointing` state. 

I would like to be able to do a normal run with POCS (for instance, with a unit set up) and have it run through a full observation cycle, even showing the fake data in PAWS.

More generally, the simulator code needs to be fixed up a bit and made consistent.

AnthonyHorton, 2018-02-13:
The camera field is only supposed to generate random data if its been asked for a dark frame. Otherwise it should just copy the data from `unsolved.fits`, generate a set of headers for it, & save the result to a FITS file in the normal location.

wtgee, 2018-02-13:
> The camera simulator is only supposed to generate random data if its been asked for a dark frame. Otherwise it should just copy the data from unsolved.fits, generate a set of headers for it, & save the result to a FITS file in the normal location.

Good to know. I was experiencing some (intermittent) issues and @jamessynge reported the same thing, so this is mostly meant as a way to explore that. While I was working on some of the DB abstraction I noticed failures in the logs that were not causing test failures.

This may be a non-issue. I'll look into what I was experiencing. 

For the record, here is what James reported:

> I'm debugging tests with my new PanMemoryDB, and in test_pocs.py am down to one failure. If I run that by itself and study the log file, I notice that the camera (simulator) doesn't seem to work right (or at least not reliably.

> So, first, what should happen with "taking" an exposure in that setting?

> And do you have any idea what the cause of the file errors below could be? If they are expected because of the simulator being incomplete, perhaps we can fix that so that it is less difficult to debug.

> Note that on my laptop, there is no /var/panoptes/images/fields directory.

```
D0209 23:58:35.321 simulator.py:79      Taking 5 second exposure on Cam00: /var/panoptes/POCS/pocs/tests/data/solved.fits
D0209 23:58:35.323 pointing.py:59       Waiting for images: 0.0 seconds
D0209 23:58:35.324 simulator.py:79      Getting mount simulator status
D0209 23:58:39.436 pointing.py:59       Waiting for images: 3.0 seconds
D0209 23:58:39.437 simulator.py:79      Getting mount simulator status
E0209 23:58:40.346 fits.py:338          Error writing image to /var/panoptes/POCS/pocs/tests/data/solved.fits!
E0209 23:58:40.352 fits.py:339          File '/var/panoptes/POCS/pocs/tests/data/solved.fits' already exists.
D0209 23:58:40.823 camera.py:239        Processing Generic PANOPTES Unit_SC2780_20160909T080000
D0209 23:58:40.823 camera.py:242        Extracting pretty image
W0209 23:58:40.823 camera.py:247        Problem with extracting pretty image: None
D0209 23:58:40.823 camera.py:467        Updating FITS headers: /var/panoptes/images/fields/Kic8462852/SC2780/20160909T080000//pointing00.fits

------------------------------------------------------------------ Captured stderr call ------------------------------------------------------------------
Exception in thread Cam00Thread:
Traceback (most recent call last):
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/threading.py", line 1182, in run
    self.function(*self.args, **self.kwargs)
  File "/home/james/git/panoptes/POCS/pocs/camera/camera.py", line 249, in process_exposure
    file_path = self._process_fits(file_path, info)
  File "/home/james/git/panoptes/POCS/pocs/camera/camera.py", line 468, in _process_fits
    fits_utils.update_headers(file_path, info)
  File "/home/james/git/panoptes/POCS/pocs/utils/images/fits.py", line 349, in update_headers
    with fits.open(file_path, 'update') as f:
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py", line 160, in fitsopen
    lazy_load_hdus, **kwargs)
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py", line 413, in fromfile
    lazy_load_hdus=lazy_load_hdus, **kwargs)
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py", line 998, in _readfrom
    fileobj = _File(fileobj, mode=mode, memmap=memmap, cache=cache)
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/utils/decorators.py", line 507, in wrapper
    return function(*args, **kwargs)
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/file.py", line 150, in __init__
    self._open_filename(fileobj, mode, overwrite)
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/file.py", line 506, in _open_filename
    self._file = fileobj_open(self.name, IO_FITS_MODES[mode])
  File "/home/james/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/astropy/io/fits/util.py", line 409, in fileobj_open
    return open(filename, mode, buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: '/var/panoptes/images/fields/Kic8462852/SC2780/20160909T080000//pointing00.fits'
```

AnthonyHorton, 2018-02-13:
OK. I see the problem. In the `take_observation()` method of the camera simulator it is calling the `_setup_observation()` method of the camera base class to generate output file paths, etc.,  the same as the real cameras do, but then it overrides the generated file path with `/var/panoptes/POCS/pocs/tests/data/solved.fits` (line 43 & 44).  This is causing it to try to write every file to the same location, resulting in file already exists errors and causing the subsequent processing to fail because that is expecting to find the file in the normal image directories.  I think all we need to do is remove those two lines and all will work as intended.

AnthonyHorton, 2018-02-13:
Must have missed those lines during my big refactor of the camera code, which did in fact make the simulator much more consistent with the real cameras in most ways.

wtgee, 2018-02-13:
Awesome, glad you saw the problem right away so I don't have to go searching. ;)

commit a37c3822 (2018-02-14): Fix camera simulator (#479)

Fixing some of the issues with the camera simulator. Thanks @AnthonyHorton!

Closes #478
closed 2018-02-14 by wtgee