panoptes/POCS #483: Add support for local fields file

jamessynge, 2018-02-14:
Just as we support reading from pocs_local.yaml, it would be nice if we could read from a local variant of the fields file. Alternately, if we could specify *additional* field file(s) in pocs_local.yaml, to augment the default fields file in pocs.yaml.

wtgee, 2018-02-14:
I thought this was already asked and answered but might have been in a private email. The fields file is pulled from the config ([here](https://github.com/panoptes/POCS/blob/develop/conf_files/pocs.yaml#L28)) so the file name can easily be overridden there. I think this is preferable to supporting the "local" mechanism.

jamessynge, 2018-02-19:
The name can not be overridden except at startup. The re-read-fields-file feature is going to read the same file name. By allowing a local variant, we then allow the user to add and remove a file (e.g. for a single night) without having to stop and restart pocs_shell twice.

wtgee, 2018-02-19:
The idea was to always integrate the scheduler into PAWS so that targets could be added or removed via a better interface, which would interact directly with the `pocs.observatory.scheduler.observations` list. A local file name could work in some scenarios but we might be better off having a conversation about the overall aims of the scheduler which to date has been very dumb.