panoptes/POCS #484: Startup script start errors

wtgee, 2018-02-15:
I get an error when it first tries to start messaging:

```
++ date
+ echo 'Running ./start_panoptes_in_tmux.sh at Wed Feb 14 15:14:30 HST 2018'
Running ./start_panoptes_in_tmux.sh at Wed Feb 14 15:14:30 HST 2018
++ pwd
+ echo 'Current dir: /var/panoptes/POCS/scripts/startup'
Current dir: /var/panoptes/POCS/scripts/startup
+ echo 'Current user: panoptes'
Current user: panoptes
+ echo 'Current path: /home/panoptes/google-cloud-sdk/bin:/var/panoptes/bin:/home/panoptes/anaconda3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/astrometry/bin:/home/panoptes/arduino-1.8.3/'
Current path: /home/panoptes/google-cloud-sdk/bin:/var/panoptes/bin:/home/panoptes/anaconda3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/astrometry/bin:/home/panoptes/arduino-1.8.3/
+ echo 'PANUSER: panoptes'
PANUSER: panoptes
+ echo 'PANDIR: /var/panoptes'
PANDIR: /var/panoptes
+ echo 'PANLOG: /var/panoptes/logs'
PANLOG: /var/panoptes/logs
+ echo 'POCS: /var/panoptes/POCS'
POCS: /var/panoptes/POCS
+ echo 'PAWS: /var/panoptes/PAWS'
PAWS: /var/panoptes/PAWS
+ echo 'PIAA: '
PIAA: 
+ echo 'MATPLOTLIBRC: '
MATPLOTLIBRC: 
+ echo '$-:' ehxB
$-: ehxB
+ set +e
+ conda info

     active environment : None
       user config file : /home/panoptes/.condarc
 populated config files : 
          conda version : 4.4.8
    conda-build version : 1.19.0
         python version : 3.5.1.final.0
       base environment : /home/panoptes/anaconda3  (writable)
           channel URLs : https://repo.continuum.io/pkgs/main/linux-64
                          https://repo.continuum.io/pkgs/main/noarch
                          https://repo.continuum.io/pkgs/free/linux-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/linux-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/linux-64
                          https://repo.continuum.io/pkgs/pro/noarch
          package cache : /home/panoptes/anaconda3/pkgs
                          /home/panoptes/.conda/pkgs
       envs directories : /home/panoptes/anaconda3/envs
                          /home/panoptes/.conda/envs
               platform : linux-64
             user-agent : conda/4.4.8 requests/2.18.4 CPython/3.5.1 Linux/4.10.6-041006-generic ubuntu/16.04 glibc/2.23
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

+ set +e
+ python /var/panoptes/POCS/pocs/utils/data.py
Downloading http://maia.usno.navy.mil/ser7/finals2000A.all [Done]
+ tmux new-window -n messaging
+ ./start_messaging_hub.sh :messaging
+ WINDOW=:messaging
++ basename ./start_messaging_hub.sh
++ date
+ echo 'Running start_messaging_hub.sh at Wed Feb 14 15:14:39 HST 2018, WINDOW=:messaging'
Running start_messaging_hub.sh at Wed Feb 14 15:14:39 HST 2018, WINDOW=:messaging
+ sleep 1s
+ tmux send-keys -t :messaging date C-m
can't find window messaging
```

jamessynge, 2018-02-19:
I've re-worked the startup scripts, but was not able to reproduce it. Could you try again after #492 is merged?

wtgee, 2018-02-19:
Sorry, this was resolved by setting the [DISABLE_AUTO_TITLE for zsh](https://github.com/robbyrussell/oh-my-zsh/issues/908).

closed 2018-02-19 by wtgee