panoptes/POCS #490: Startup script changes

wtgee, 2018-02-16:
Startup script has been helpful but here are just a few things based on usage:

* Impatient people like me want to attach to the session immediately but then I tend to start typing keys and this tends to screw things up. Maybe put a sleep timer at end of `tmux_launch` that says when it is okay to attach (and how). Or maybe just attach?
* I don't think the peas_shell log is very helpful to see and it also is the default screen that is shown when you attach. At a minimum we should launch the peas log first then the pocs log so that it shows pocs log when attaching. But I mostly think we just don't need to follow the peas log. Or maybe place the next to each other. So launch peas, then log file then launch pocs, then log file.

jamessynge, 2018-02-16:
We'll need to add grc to the install script.

Do you still want to switch to byobu?

jamessynge, 2018-02-19:
@wtgee Can you provide a grc config file that does some useful colorizing? I tried it on PAN006, and didn't see much benefit.

wtgee, 2018-02-19:
I already posted one in #491. It doesn't provide it out of the box because we alter our log format to be different than what is standard.

jamessynge, 2018-02-19:
Sorry, didn't see that. Repeating here:

```
# PANOPTES log format coloring
#
# Dates
regexp=[^\d]*\d\d:\d\d:\d\d[\.,]{0,1}\d{0,1}\d{0,1}\d{0,1}
colours=bold blue
count=once
======
# DEBUG
regexp=^D.*$
colours=bold default
======
# INFO
regexp=^I.*$
colours=bold blue
======
# WARNING
regexp=^W.*$
colours=yellow
======
# ERROR
regexp=^E.*$
colours=bold red
======
# CRITICAL
regexp=^C.*$
colours=bold on_red
======
```

jamessynge, 2018-03-25:
Regarding:
> I don't think the peas_shell log is very helpful to see and it also is the default screen that is shown when you attach. At a minimum we should launch the peas log first then the pocs log so that it [...]

I addressed this in #492 by re-ordering the windows as follows:

1. Messaging Hub
2. PAWS
3. PEAS log file viewer
4. POCS log file viewer
5. PEAS shell
6. POCS shell

POCS shell becomes the current window when it is created, so the first attach operation will see that window.

jamessynge, 2018-03-25:
Regarding:
> Impatient people like me want to attach to the session immediately but then I tend to start typing keys and this tends to screw things up. Maybe put a sleep timer at end of tmux_launch that says when it is okay to attach (and how). Or maybe just attach?

I don't think a timer will help because the user won't see any message emitted by tmux launch unless watching the log file. And I don't understand what "just attaching" would mean.

We could modify the script to create all of the windows, including an extra window that includes a message saying that startup is still in progress. When startup gets to the end, it would delete that window. 

We could also provide a script to be executed from .profile which (in interactive shells) prints a message about the status of start_panoptes_in_tmux.sh, initially saying:
> start_panoptes_in_tmux is in progress, please wait before attaching to the panoptes session.
> Execute `$POCS/scripts/panoptes_status.sh` to determine the status of the PANOPTES software.

At the end of the script, it could modify what the script says to:
> Execute `tmux attach-session -t panoptes`"

And it can execute this command to indicate that it done:
> `wall "PANOPTES software is running. Execute 'tmux attach-session -t panoptes' to take control."`

Note though that `wall` doesn't work with gnome-terminal.

Thoughts?

wtgee, 2018-08-27:
@jamessynge there have been other issues updating the startup script since this issue. Should we close here?

wtgee, 2018-10-16:
Closing as stale.

closed 2018-10-16 by wtgee