panoptes/POCS #494: Add timelapse legend to images/fields/... directory

jamessynge, 2018-02-19:
I want things both ways: to be able to look at a bunch of timelapse movies all at once (i.e. by looking at the images/timelapse directory), and also when looking at a particular observation folder (e.g. /var/panoptes/images/fields/HatP20/7bab97/20180218T231235). A benefit of the latter is that we can then easily uploaded it to GCS, making it easier for folks to get a quick summary of the observation by viewing the timelapse.

wtgee, 2018-02-19:
I think this would be best as a functionality of PAWS.

jamessynge, 2018-02-21:
Sorry, I copied the title from the previous issue, and forgot to remove the word 'legend'. The primary point here is that the .mp4 timelapse files should be uploaded along with the .fits.fz image files.

Secondarily, I'm saying that I want it to be easy to find the timelapse images for the current/previous night. That could well be a feature of PAWS.

wtgee, 2018-02-21:
I've gone back and forth about keeping the timelapses in a storage bucket as I think they have very limited utility and, ultimately, could be recreated from the data if needed. Usually they will be viewed for the previous night or one or two nights afterward but probably never again so I think they are okay to just store locally and make available via PAWS. Could in the future have some sort of share feature maybe?

wtgee, 2022-01-15:
Stale.

closed 2022-01-15 by wtgee