panoptes/POCS #495: PAN006 stopped observing ... due to tracking correction error?

jamessynge, 2018-02-19:
Before midnight last night PAN006 stopped observing. I had set it up to observe 55 Cnc as its only target, as there was a transit of 55 Cnc e due between midnight and 2am.

Looking at pocs_shell-all.log (see fragment below), it was 5 exposures into an observation run of 60 exposures when there was some problem with tracking. The generated timelapse images (4 observations of the field) show that the first field looks NOTHING like the 2nd, 3rd and 4th (which look to be the same). And while the tracking was very stable for the first field, it was shifting quite a bit during the other 3 observations.

```
=<Angle -348.26369688000227 arcsec>, magnitude=<Angle 348.296330752448 arcsec>)
D0219 04:29:15.745 machine.py:267       After calling analyze. Now in analyzing state
D0219 04:29:15.747 machine.py:309       Source: analyzing        Dest: tracking
D0219 04:29:15.747 machine.py:171       Transition method: adjust_tracking
D0219 04:29:15.747 machine.py:209       Checking safety for adjust_tracking
D0219 04:29:15.835 core.py:336          Dark Check: True
D0219 04:29:15.835 core.py:351          Checking weather safety
D0219 04:29:15.836 core.py:371          Weather Safety: True [66 sec old - 2018-02-19 04:28:10.282000]
D0219 04:29:17.850 machine.py:253       Before calling adjust_tracking from analyzing state
I0219 04:29:22.476 messaging.py:134     PANCHAT Checking our tracking
D0219 04:29:22.476 observatory.py:342   Updating the tracking
I0219 04:29:22.478 observatory.py:372   Adjusting Dec: south 69462.48 ms -348.26 arcsec
D0219 04:29:26.994 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:29.113 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:31.233 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:33.352 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:35.469 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:37.586 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:39.703 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:41.821 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:43.938 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:46.050 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:48.167 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:50.288 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:52.401 observatory.py:383   Waiting for Dec tracking adjustment
D0219 04:29:54.512 observatory.py:383   Waiting for Dec tracking adjustment
W0219 04:29:56.628 tracking.py:15       Problem adjusting tracking: Trying to adjust Dec tracking for more than 30 se
conds
D0219 04:29:56.629 machine.py:267       After calling adjust_tracking. Now in tracking state
D0219 04:29:56.631 machine.py:309       Source: tracking         Dest: parking
D0219 04:29:56.632 machine.py:171       Transition method: park
D0219 04:29:56.632 machine.py:209       Checking safety for park
D0219 04:29:56.632 machine.py:214       Always safe to move to park
D0219 04:29:56.632 machine.py:253       Before calling park from tracking state
D0219 04:30:01.472 observation.py:141   Resetting observation 55 Cnc e: 60.0 s exposures in blocks of 10, minimum 60, priority 300
I0219 04:30:01.472 scheduler.py:116     Setting new observation to None
I0219 04:30:01.473 messaging.py:134     PANCHAT Closing dome
D0219 04:30:01.473 abstract_serial_dome.py:70 Already connected to dome
```

jamessynge, 2018-02-19:
For the first image of the first observation, solve_field.sh has this solution:

```
Field 1 solved: writing to file ./20180219T021754.solved to indicate this.
Field: 20180219T021754.fits
Field center: (RA,Dec) = (133.934741, 30.204882) deg.
Field center: (RA H:M:S, Dec D:M:S) = (08:55:44.338, +30:12:17.576).
Field size: 14.8174 x 9.91915 degrees
Field rotation angle: up is -90.4891 degrees E of N
Field parity: pos
Creating new FITS file "./20180219T021754.new"...
```
The first field of the 4th observation has this solution:
```
Field 1 solved: writing to file ./20180219T041718.solved to indicate this.
Field: 20180219T041718.fits
Field center: (RA,Dec) = (135.030538, 26.529858) deg.
Field center: (RA H:M:S, Dec D:M:S) = (09:00:07.329, +26:31:47.488).
Field size: 14.8191 x 9.92006 degrees
Field rotation angle: up is 90.0776 degrees E of N
Field parity: pos
Creating new FITS file "./20180219T041718.new"...
```
From the rotation angle I gather that the meridian flip occurred after the first observation. I'm unclear on how the decision to do the meridian flip is made; by the mount?

wtgee, 2018-02-19:
The mount should be stopping when it gets to 30 degrees past the meridian,
so an automatic meridian flip shouldn't have occurred. I had emailed you
instructions on Jan 21 when you were going to Wheaton so I'm pretty sure
you had implemented this but it looks like maybe not. We don't want to keep
observing through a meridian flip and any such flip should be considered a
new observation.

However, I'm curious as to why it even scheduled the target if it knew it
had an upcoming meridian flip. You left the min_nexp at 60? I suppose if
the settings didn't get changed it would flip at 0°exactly whereas the code
assumes it can go to 30°past.

Here's also where the retries should have come into play. After the parking
from bad tracking correction did it try again?

jamessynge, 2018-02-20:
I did make the meridian flip change back in January, but can't prove it (i.e. I don't think I can read the setting form the mount). I cc'd you on an email to iOptron about this.

I didn't specify min_nexp, just this:
```yaml
# Targets for 2018-02-18, which have transits.
# 55 Cnc e
-
    name: 55 Cnc e
    position: 08h52m36.13s +28d19m53.0s
    priority: 300
    exp_time: 60
```
No, it didn't try again after the tracking failure, it gave up for the night. I can't tell if perhaps it thought that 55 Cnc was too low in the sky. I'll send you the full log.

jamessynge, 2018-02-20:
Does POCS have the ability to control when the meridian flip occurs? Does it occur implicitly when slewing to a _new_ target (e.g. go home, go back to target)?

wtgee, 2018-02-20:
No, we don't have explicit control over the meridian flip and yes it does
occur implicitly with a new target (no "go home" required right now).

wtgee, 2018-02-20:
Odd to me that it wouldn't try again. I'll check out the full logs, thanks.

commit 709d7f7e (2018-08-19): Tracking error fixes (#549)

* Fixing direction depending on where pier is pointed.

Closes #495
closed 2018-08-19 by wtgee