panoptes/POCS #498: Add WCS information to fits header

rzellem, 2018-02-22:
Adding WCS information (http://docs.astropy.org/en/stable/wcs/) to the fits header will be very helpful when reducing the data as it will allow the user/code to quickly find the target of interest from its RA+DEC rather than its pixel location. Particularly helpful in large fields with dim target stars and for fields where the target has shifted its location on the detector due to, for example, a meridian flip.

wtgee, 2018-02-22:
Each unit has one camera that is designated as the `primary` camera and we do plate-solving on those images while the unit is running (we implement a simple closed-loop tracking correction). So if you check both cameras for a given observation one set of them should already have WCS.

We had planned to solve the other set during the pipeline process but it might make more sense to do it during the `housekeeping` state at the end of the day so that all images that are uploaded have a WCS.

Note that we never do a meridian flip during a course of a single Observation. Essentially once the target has shifted pixels in any significant way we want to start a new Observation, even if it is on the same target/field. This is due to how the algorithm works for processing the RGB array. Observations are all contained within one folder and are also known as a "sequence" (with a corresponding `SEQID` in the fits header).

wtgee, 2018-02-22:
Immediate follow-up: I don't see any reason not to do this during `housekeeping` so let's go with that.

rzellem, 2018-02-22:
Ah, ok, then I was probably looking at data from the other camera that didn't have WCS information. Sorry, still learning how the data is stored into the subdirectories.

Quick follow up question: how does the location of the target on the chip affect the RGB array processing? Aren't the R, G, and B pixels always fixed on the array and therefore should always be extracted in a similar way?

wtgee, 2020-02-17:
Closing as stale.

closed 2020-02-17 by wtgee