panoptes/POCS #505: pocs_shell loop stops at end of night

jamessynge, 2018-03-12:
At dawn this morning POCS went through the end of night cleanup, but then printed this:

```
I0311 10:21:21.852 messaging.py:134     PANCHAT Weather is good and it is dark. Something must have gone wrong. Stopping loop.
```

The previous Dark Check reported False (see below), so I don't know why it would be reporting that it is dark.

Another problem is that it has apparently stopped the `run_pocs` loop (the logging stopped shortly after that), but pocs_shell didn't not prompt for another command.

```
D0311 10:12:27.985 core.py:336          Dark Check: True
D0311 10:12:27.985 core.py:351          Checking weather safety
D0311 10:12:27.986 core.py:371          Weather Safety: True [48 sec old - 2018-03-11 10:11:40.474000]
D0311 10:12:30.002 machine.py:253       Before calling track from pointing state
D0311 10:12:34.619 machine.py:267       After calling track. Now in tracking state
D0311 10:12:34.622 machine.py:309       Source: tracking	 Dest: observing
D0311 10:12:34.622 machine.py:171       Transition method: observe
D0311 10:12:34.622 machine.py:209       Checking safety for observe
D0311 10:12:34.730 core.py:336          Dark Check: True
D0311 10:12:34.730 core.py:351          Checking weather safety
D0311 10:12:34.732 core.py:371          Weather Safety: True [54 sec old - 2018-03-11 10:11:40.474000]
D0311 10:12:36.747 machine.py:253       Before calling observe from tracking state
I0311 10:12:41.651 messaging.py:134     PANCHAT I'm finding exoplanets!
D0311 10:12:41.652 observatory.py:425   Getting headers for : KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
D0311 10:12:42.266 observatory.py:263   Exposing for camera: Cam00
D0311 10:12:42.266 camera.py:437        image_id: Panoptes Unit PAN006 @ Wheaton College_6575fc_20180311T101242
D0311 10:12:42.266 canon_gphoto2.py:124 Taking 120.0 second exposure on Cam00: /var/panoptes/images/fields/Kic8462852/6575fc/20180311T101114//20180311T101242.cr2
D0311 10:12:42.272 observatory.py:263   Exposing for camera: Cam01
D0311 10:12:42.273 camera.py:437        image_id: Panoptes Unit PAN006 @ Wheaton College_7bab97_20180311T101242
D0311 10:12:42.273 canon_gphoto2.py:124 Taking 120.0 second exposure on Cam01: /var/panoptes/images/fields/Kic8462852/7bab97/20180311T101114//20180311T101242.cr2
D0311 10:12:42.279 observing.py:25      Waiting for images: 0.0 seconds
D0311 10:13:01.672 observing.py:25      Waiting for images: 15.0 seconds
D0311 10:13:21.679 observing.py:25      Waiting for images: 30.0 seconds
D0311 10:13:41.669 observing.py:25      Waiting for images: 45.0 seconds
D0311 10:14:01.691 observing.py:25      Waiting for images: 60.0 seconds
D0311 10:14:21.672 observing.py:25      Waiting for images: 75.0 seconds
D0311 10:14:41.691 observing.py:25      Waiting for images: 90.0 seconds
D0311 10:14:48.272 camera.py:241        Processing Panoptes Unit PAN006 @ Wheaton College_6575fc_20180311T101242
D0311 10:14:48.272 camera.py:244        Extracting pretty image
D0311 10:14:48.278 canon_gphoto2.py:154 Converting CR2 -> FITS: /var/panoptes/images/fields/Kic8462852/6575fc/20180311T101114//20180311T101242.cr2
D0311 10:14:48.283 camera.py:241        Processing Panoptes Unit PAN006 @ Wheaton College_7bab97_20180311T101242
D0311 10:14:48.284 camera.py:244        Extracting pretty image
D0311 10:14:48.295 canon_gphoto2.py:154 Converting CR2 -> FITS: /var/panoptes/images/fields/Kic8462852/7bab97/20180311T101114//20180311T101242.cr2
D0311 10:14:51.774 camera.py:254        Adding current observation to db: Panoptes Unit PAN006 @ Wheaton College_6575fc_20180311T101242
D0311 10:14:51.776 camera.py:263        Adding image metadata to db: Panoptes Unit PAN006 @ Wheaton College_6575fc_20180311T101242
D0311 10:14:51.855 camera.py:260        Compressing /var/panoptes/images/fields/Kic8462852/7bab97/20180311T101114//20180311T101242.fits
D0311 10:14:52.091 camera.py:263        Adding image metadata to db: Panoptes Unit PAN006 @ Wheaton College_7bab97_20180311T101242
D0311 10:15:01.713 observing.py:42      Finished with observing, going to analyze
D0311 10:15:01.713 machine.py:267       After calling observe. Now in observing state
D0311 10:15:01.716 machine.py:309       Source: observing	 Dest: analyzing
D0311 10:15:01.716 machine.py:171       Transition method: analyze
D0311 10:15:01.716 machine.py:209       Checking safety for analyze
D0311 10:15:01.980 observatory.py:82    Sun -9.82 deg > -10.0 deg
D0311 10:15:01.980 core.py:336          Dark Check: False
D0311 10:15:01.981 core.py:351          Checking weather safety
D0311 10:15:01.982 core.py:371          Weather Safety: True [36 sec old - 2018-03-11 10:14:26.411000]
W0311 10:15:01.983 core.py:308          Unsafe conditions: {'is_dark': False, 'good_weather': True, 'free_space': True}
W0311 10:15:01.983 core.py:311          Safety failed so sending to park
D0311 10:15:01.983 machine.py:209       Checking safety for park
D0311 10:15:01.983 machine.py:214       Always safe to move to park
D0311 10:15:01.983 machine.py:253       Before calling park from observing state
D0311 10:15:06.697 observation.py:141   Resetting observation KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
I0311 10:15:06.697 scheduler.py:116     Setting new observation to None
I0311 10:15:06.698 messaging.py:134     PANCHAT Closing dome
D0311 10:15:06.699 abstract_serial_dome.py:70 Already connected to dome
I0311 10:15:06.951 observatory.py:536   Closed dome
D0311 10:15:14.022 astrohaven.py:183    Got target_feedback, feedback_countdown=1
D0311 10:15:14.432 astrohaven.py:183    Got target_feedback, feedback_countdown=0
D0311 10:15:21.802 astrohaven.py:183    Got target_feedback, feedback_countdown=1
[... 65 lines cut ...]
D0311 10:19:45.532 observatory.py:221   Housekeeping for Kepler 1100: 120.0 s exposures in blocks of 10, minimum 60, priority 125
D0311 10:19:45.532 observatory.py:225   Cleanup for camera Cam00 [6575fc]
D0311 10:20:23.403 observatory.py:237   Uploading directory to google cloud storage
D0311 10:20:39.101 observatory.py:225   Cleanup for camera Cam01 [7bab97]
D0311 10:20:54.918 observatory.py:237   Uploading directory to google cloud storage
D0311 10:21:10.480 observatory.py:240   Cleanup finished
D0311 10:21:10.480 observatory.py:221   Housekeeping for KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
D0311 10:21:10.480 observatory.py:225   Cleanup for camera Cam00 [6575fc]
D0311 10:21:11.841 observatory.py:237   Uploading directory to google cloud storage
D0311 10:21:14.273 observatory.py:225   Cleanup for camera Cam01 [7bab97]
D0311 10:21:14.665 observatory.py:237   Uploading directory to google cloud storage
D0311 10:21:17.123 observatory.py:240   Cleanup finished
D0311 10:21:17.123 scheduler.py:202     Resetting observed list
I0311 10:21:17.125 messaging.py:134     PANCHAT Ok, I'm done cleaning up all the recorded data.
D0311 10:21:17.125 machine.py:267       After calling clean_up. Now in housekeeping state
D0311 10:21:17.129 machine.py:309       Source: housekeeping	 Dest: sleeping
D0311 10:21:17.129 machine.py:171       Transition method: goto_sleep
D0311 10:21:17.129 machine.py:209       Checking safety for goto_sleep
D0311 10:21:17.129 machine.py:214       Always safe to move to goto_sleep
D0311 10:21:17.129 machine.py:253       Before calling goto_sleep from housekeeping state
I0311 10:21:21.852 messaging.py:134     PANCHAT Weather is good and it is dark. Something must have gone wrong. Stopping loop.
I0311 10:21:21.852 machine.py:181       Stopping POCS states
I0311 10:21:21.853 messaging.py:134     PANCHAT Another successful night!
D0311 10:21:21.853 machine.py:267       After calling goto_sleep. Now in sleeping state
```

jamessynge, 2018-07-06:
See also #527 and #504.

wtgee, 2020-06-14:
Closing as stale.

closed 2020-06-14 by wtgee