panoptes/POCS #508: pocs_shell exit hanging

jamessynge, 2018-03-27:
The log file showed no activity after I entered exit, but pocs_shell did not stop. It looks like it might have been waiting 30 minutes for the sky to be clear... in part because it hadn't detected that I'd changed pocs_local.yaml to point to a new targets file.

This is the output in the terminal:

```
POCS > run_pocs                                                                                                     
Starting POCS - Press Ctrl-c to interrupt
^CPOCS interrupted, skipping states and parking
POCS stopped.
POCS > exit
Shutting down POCS instance, please wait
^CTraceback (most recent call last):
  File "bin/pocs_shell", line 899, in <module>
    PocsShell().cmdloop()
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/cmd.py", line 138, in cmdloop
    stop = self.onecmd(line)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/cmd.py", line 217, in onecmd
    return func(arg)
  File "bin/pocs_shell", line 256, in do_exit
    self.do_power_down()
  File "bin/pocs_shell", line 332, in do_power_down
    self.pocs.power_down()
  File "/var/panoptes/POCS/pocs/core.py", line 243, in power_down
    self.set_park()
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 382, in trigger
    return self.machine._process(func)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 1007, in _process
    return trigger()
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 398, in _trigger
    return self._process(event_data)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 408, in _process
    if trans.execute(event_data):
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 262, in execute
    self._change_state(event_data)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 273, in _change_state
    event_data.machine.get_state(self.dest).enter(event_data)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 119, in enter
    event_data.machine.callback(handle, event_data)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/transitions/core.py", line 988, in callback
    func(event_data)
  File "/var/panoptes/POCS/pocs/state/states/default/parked.py", line 28, in on_enter
    pocs.sleep(delay=1800)  # 30 minutes = 1800 seconds
  File "/var/panoptes/POCS/pocs/core.py", line 430, in sleep
    time.sleep(10.0)
KeyboardInterrupt
--- Logging error ---
Traceback (most recent call last):
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/handlers.py", line 73, in emit
    logging.FileHandler.emit(self, record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 1069, in emit
    self.stream = self._open()
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 1059, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
NameError: name 'open' is not defined
Call stack:
  File "/var/panoptes/POCS/pocs/dome/abstract_serial_dome.py", line 42, in __del__
    self.serial.disconnect()
  File "/var/panoptes/POCS/pocs/utils/rs232.py", line 156, in disconnect
    self.logger.debug('SerialData.disconnect called for {}', self.name)
Message: 'SerialData.disconnect called for {}'
Arguments: ('/dev/ttyUSB2',)
(panoptes-env) panoptes@Panmass:/var/panoptes/POCS ((30a2828...))$ 
```

And this is the end of the log file:

```
E0326 23:54:37.009 error.py:18          NoObservation: No valid observations found
I0326 23:54:37.011 messaging.py:134     PANCHAT No valid observations found. Can't schedule. Going to park.
D0326 23:54:37.012 machine.py:267       After calling schedule. Now in scheduling state
[... 133 lines cut ...]
D0327 00:27:49.565 ioptron.py:322       RA (ms): 69640548.92458375 ms
D0327 00:27:49.565 ioptron.py:325       Dec (centiarcsec): -3600000.0 carcsec
D0327 00:27:51.068 ioptron.py:187       Slewing to park
D0327 00:28:11.653 mount.py:460         Mount unparked
D0327 00:28:12.154 mount.py:474         Move command: move_south
D0327 00:28:12.155 mount.py:478         Moving south for 11.0 seconds. 
D0327 00:28:25.672 mount.py:483         13.51504099999481 seconds passed before stop
D0327 00:28:26.175 mount.py:485         14.018590999999248 seconds passed total
D0327 00:28:26.175 mount.py:493         Stopping movement
D0327 00:28:28.690 mount.py:368         Mount parked
I0327 00:28:45.094 messaging.py:134     PANCHAT I'm powering down
I0327 00:28:45.095 core.py:225          Shutting down POCS State Machine: , please be patient and allow for exit.
D0327 00:28:45.095 abstract_serial_dome.py:70 Already connected to dome
D0327 00:28:45.522 astrohaven.py:183    Got target_feedback, feedback_countdown=1
D0327 00:28:46.934 astrohaven.py:183    Got target_feedback, feedback_countdown=0
D0327 00:28:47.842 astrohaven.py:183    Got target_feedback, feedback_countdown=1
D0327 00:28:49.255 astrohaven.py:183    Got target_feedback, feedback_countdown=0
I0327 00:28:52.152 core.py:242          Mount is parked, setting Parked state
D0327 00:28:52.152 machine.py:209       Checking safety for set_park
D0327 00:28:52.152 machine.py:214       Always safe to move to set_park
D0327 00:28:52.153 machine.py:253       Before calling set_park from parking state
I0327 00:28:56.859 messaging.py:134     PANCHAT I'm parked now. Phew.
I0327 00:28:56.860 messaging.py:134     PANCHAT No observations found.
I0327 00:28:56.861 messaging.py:134     PANCHAT Going to stay parked for half an hour then will try again.
```

wtgee, 2018-05-07:
Tracing this down, it looks like it is giving an error while trying to write to the log file in the `rs232.disconnect()` method. Would the `__del__` be called after the log file is closed? @AnthonyHorton saw something similar with the SBIG cameras and the `__del__` dunder and eventually just removed the logging call.

@jamessynge You said this hung for 30 minutes after typing `exit`?

wtgee, 2019-08-25:
Closing as stale.

closed 2019-08-25 by wtgee