panoptes/POCS #522: Output the `pocs.say` messages to Twitter (and possibly more)

jeremylan, 2018-06-01:
This feature was suggested by @jamessynge on the forums, see the discussion here:

https://forum.projectpanoptes.org/t/help-getting-started/71

Something that is not critical but will be really useful would be the ability to send a tweet for each of the “say” messages that the software posts; these are intended for human consumption, such as “Parking the mount” or “Observing HAT-9”.

I created a simple Python script that uses the `PanMessaging` class to subscribe to the `PANCHAT` channel.

I used the Tweepy library (http://www.tweepy.org/) to output the messages on a Twitter account I have for development.

It would be good now to get a little more requirements for me to keep working at this i.e.

- stuff to put on a config file (i.e. Twitter API keys, channels to listen to, etc.)
- timeouts and how to create this process, i.e. an executable, a simple script to run via Python...
- should I also consider outputting to a Slack account?

Any suggestion welcome
Thanks

wtgee, 2018-06-01:
Awesome, thanks @jeremylan! Great that you are jumping in.

For another project that subclasses POCS I wanted to get the slack feature working. Turns out it was dead simple. I haven't yet put out a PR for that but you can see it [here](https://github.com/panoptes/POCS/blob/may-run/pocs/core.py#L185).

It would definitely be a little cleaner if it were hidden away within some kind of `PanMessaging` class as I'm imagining you've done. You could go ahead and put that slack code listed above alongside whatever you have done, which would allow you to test how to implement it in two different ways, which is nice. We're always happy to have a PR coming in!

The above code just assumes there is a `slack_webhook_url` in the config but obviously that will start to expand as we add things like twitter, youtube (timelapse movie upload), etc. We should create a  top-level config item (maybe `social_accounts`?) with sub-items such as api keys, url, etc. 

Just a note, the config files are in `$POCS/conf_files`, with the main file being `pocs.yaml`. The loading of the config automatically supports a localized version, so anything in `$POCS/conf_files/pocs_local.yaml` will override the default config items.

Alternatively we could just have a separate config file (e.g. `$POCS/conf_files/social.yaml` - which also automatically supports `social_local.yaml`). This would then require a slight modification to how we construct a [`POCS`](https://github.com/panoptes/POCS/blob/develop/pocs/core.py#L19) instance by doing something like `pocs = POCS(config=['social'])` in the `pocs_shell`, but has the advantage of keeping the main config file clean of these somewhat optional features.

wtgee, 2018-06-01:
#28

jeremylan, 2018-06-02:
Hi @wtgee thanks for the reply

That's great, I'll work at my solution to integrate with the config files as you suggest and add both Twitter and Slack.

I'll then send you the PR for review, I'm just learning the system (& Python) so please bear with me, I am also working at this on my spare time so it may take a little while :)

If you have time, could you please have a quick look at the core dump error I get on my POCS installation? It only runs for a few minutes before quitting. Error details are on the forum thread.

Talk soon and thank you

wtgee, 2018-06-06:
Awesome, thanks @jeremylan ! I'll take a look at the PR. The logging is probably related to #393, which has been hard to pin down. I was doing some work in the last few days where I was seeing this as well. Something related to loading logging from an outside module before loading POCS. The `fileline` it refers to is the custom log formatting we do.

I'll get to PR in a few hours. Cheers!

jeremylan, 2018-06-06:
Ahhh yes @wtgee, definitely same issue as #393.

Glad it's not something I introduced ;P

closed 2018-07-01 by jamessynge