panoptes/POCS #526: Remove utc_offset

wtgee, 2018-06-29:
Determine if `utc_offset` is used anywhere and if not remove from config.

jamessynge, 2019-03-09:
FWIW, I see that utc_offset is an attribute that [astropy uses in its timezone info](http://docs.astropy.org/en/stable/search.html?q=utc_offset&check_keywords=yes&area=default). Not sure if that matters for us or not.

wtgee, 2019-03-09:
> FWIW, I see that utc_offset is an attribute that [astropy uses in its timezone info](http://docs.astropy.org/en/stable/search.html?q=utc_offset&check_keywords=yes&area=default). Not sure if that matters for us or not.

We don't appear to be using it at all.

ImmortalRabbit, 2019-03-19:
@wtgee I think this issue should be closed because my solution seems like correct and you have already merged it. Or are there any other things to do?

wtgee, 2019-03-19:
> @wtgee I think this issue should be closed because my solution seems correct and you have already merged it. Or are there any other things to do?

All good, thanks! FYI, in your PR you have "Fixes issue #526" but if you put "Fixes #526" it would have automatically closed the issue upon merge. See [here](https://help.github.com/en/articles/closing-issues-using-keywords) for more example.

closed 2019-03-19 by wtgee

ImmortalRabbit, 2019-03-20:
@wtgee thanks for suggestion, next time I will try to do not repeat this kind of mistake.