panoptes/POCS #527: POCS stuck

jamessynge, 2018-06-30:
No observations by PAN006 for the last 9 days because POCS got confused during cleanup. Is this happening on PAN001?

This is not the first time this has happened. Makes me think that we might well want to have a noon restart daily. Thoughts?

```
D0620 07:53:45.870 observatory.py:238   Uploading directory to google cloud storage
D0620 07:53:58.708 observatory.py:241   Cleanup finished
D0620 07:53:58.708 observatory.py:222   Housekeeping for KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
D0620 07:53:58.708 observatory.py:226   Cleanup for camera Cam00 [6575fc]
D0620 07:54:01.976 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:04.990 observatory.py:226   Cleanup for camera Cam01 [7bab97]
D0620 07:54:06.424 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:09.474 observatory.py:241   Cleanup finished
D0620 07:54:09.474 observatory.py:222   Housekeeping for KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
D0620 07:54:09.474 observatory.py:226   Cleanup for camera Cam00 [6575fc]
D0620 07:54:12.505 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:15.149 observatory.py:226   Cleanup for camera Cam01 [7bab97]
D0620 07:54:16.361 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:19.021 observatory.py:241   Cleanup finished
D0620 07:54:19.021 observatory.py:222   Housekeeping for KIC 8462852: 120.0 s exposures in blocks of 10, minimum 60, priority 200
D0620 07:54:19.021 observatory.py:226   Cleanup for camera Cam00 [6575fc]
D0620 07:54:22.130 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:24.722 observatory.py:226   Cleanup for camera Cam01 [7bab97]
D0620 07:54:26.000 observatory.py:238   Uploading directory to google cloud storage
D0620 07:54:28.514 observatory.py:241   Cleanup finished
D0620 07:54:28.515 scheduler.py:202     Resetting observed list
I0620 07:54:28.516 messaging.py:134     PANCHAT Ok, I'm done cleaning up all the recorded data.
D0620 07:54:28.516 machine.py:267       After calling clean_up. Now in housekeeping state
D0620 07:54:28.518 machine.py:309       Source: housekeeping     Dest: sleeping
D0620 07:54:28.518 machine.py:171       Transition method: goto_sleep
D0620 07:54:28.519 machine.py:209       Checking safety for goto_sleep
D0620 07:54:28.519 machine.py:214       Always safe to move to goto_sleep
D0620 07:54:28.519 machine.py:253       Before calling goto_sleep from housekeeping state
I0620 07:54:33.291 messaging.py:134     PANCHAT Weather is good and it is dark. Something must have gone wrong. Stopping loop.
I0620 07:54:33.291 machine.py:181       Stopping POCS states
I0620 07:54:33.292 messaging.py:134     PANCHAT Another successful night!
D0620 07:54:33.292 machine.py:267       After calling goto_sleep. Now in sleeping state
```

wtgee, 2018-06-30:
The error is happening some time before the cleanup. It is then probably going through a number of the retries. Those retries should be getting reset every day but apparently aren't. Or you had a number of retries in a night. Can you post the full log for that run?

The restart idea has been discussed but I think we were waiting until after finishing all the dependency injection into the `Observation`. The idea was to keep some simple daemon (a `POCS` instance?) but to shut down everything else. This includes power down of components for the day.

A simple restart might be a good first step but I'd also like to figure out why it's happening and prevent it. However, it wouldn't be too hard to have a simple option that shuts it down in the day (like the `run_once` param).

jamessynge, 2018-07-04:
FWIW, I added a daily cron job to PAN006 that checks whether POCS has gotten stuck (ceased logging). If so, it reboots the system.

I agree that it would be great to fully diagnose this, but keeping the system up after it has gotten stuck isn't helping with that AFAICT.

wtgee, 2018-08-27:
@jamessynge can you post details about the cron job somewhere?

wtgee, 2019-08-25:
Closing as stale.

closed 2019-08-25 by wtgee