panoptes/POCS #531: Create script for full setup of computer for a unit

jamessynge, 2018-07-08:
I've created various scripts and a document to assist in setting up the computer that runs a unit, but the process isn't yet simple (see #115). There are too many steps that the user must do. I propose that we create a script that is used immediately after Linux has been installed, wifi or ethernet configured and the owner's account (**not** `panoptes`) has been created. The script will require internet access to download dependencies (and itself), and will be executed like this:

```bash <(curl -s https://raw.githubusercontent.com/panoptes/POCS/develop/scripts/install/install-panoptes.sh)```

This script might trigger other scripts to be downloaded and executed.

I propose that the script do the following, specifically targeting Ubuntu 18.04 only for now:

- Present a set of environment variable values (e.g. PANUSER, PANDIR, etc.) and their default values
- Indicate if the user or directories already exists (esp. the user as the script will overwrite some of the files in the home dir).
- Allow the user to override the env vars.
- IFF user $PANUSER already exists, tell the user to delete the user before running this script again. The alternative is to complicate this script with lots of testing for already installed packages, etc.
- Create user $PANUSER. This will prompt for a password. Need to provide a suggestion that the user record this somewhere safe, for example a password vault such as LastPass.
- Add user $PANUSER to all of the important groups (e.g. adm, sudo, dialup, plugdev) so that it will have the ability to install software and access devices).
- Create the directories specified by the env vars, owned by user $PANUSER.
- Install anaconda for user $PANUSER, ideally without prompting, which I think the anaconda installer does by default.
- Install various software packages using conda (preferably) or pip (if not supported by conda) or directly (if supported by neither, such as mongodb or git).
- Replace the .profile and .bashrc of $PANUSER so that all shells run with the desired env vars, and with the panoptes anaconda environment.
- Clone the github repos and use setup tools to install them (e.g. python $POCS/setup.py install) and then to test them once they are all installed.

Assigning to @wtgee for review of the proposal. I'll proceed with doing some implementation.

wtgee, 2018-07-11:
This sounds good, thanks. A few quick comments:

* I think we can allow the user to override defaults but encourage them not to unless they have a good reason. With the separate user it should be a lot cleaner to just leave as is.
* anaconda needs a flag to auto-accept the EULA otherwise it hangs.
* Use miniconda over anaconda.
* Should probably make $USER a part of the $PANUSER group and make $PANDIR group r/w.
* Setup the cron jobs (at least the data download one)
* Instead of cloning do we want to just download a zip file? Could avoid git altogether if we assume they are not developing and want atomic updates.

wtgee, 2018-07-11:
Do we all need an `update-dependencies.sh` at some point? If they want to update POCS are they expected to rerun this? That changes how the user creation and warning should work by default. It might not be enough to tell them to do a git pull (assuming not using zip as above) since it might also require new dependencies at some point.

wtgee, 2020-02-16:
This has changed a lot with the addition of the docker setup. See #951.

PR #859 (2019-10-13): Utils and docker
**Breaking Changes**

This will move POCS over to a Docker based system, utilizing the panoptes-utils module

Will replace #851 and #778 

## Description

## Related Issue
#851 #778 #850 #734 

Closes #531 
Closes #519 
Closes #238 
Closes #555 
Closes #137 
Closes #94
PR #951 (2020-06-01): Docker into develop 02
Merging the `docker` branch into `develop`.

This PR is effectively `v0.7.0`

## Related Issue
Replaces #950 
Closes #927 
Closes #934 
Closes #948 
Closes #531 
Closes #137
Closes #734 
Closes #960
Closes #828 
Closes #952 
Closes #897 
Closes #809 
Closes #970 
Related to #115 

Relies on https://github.com/panoptes/panoptes-utils/pull/189

## Types of changes

- [x] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to change)
closed 2020-06-01 by wtgee