panoptes/POCS #534: missing import in setup script ?

gnthibault, 2018-07-10:
Here is what I got after installing panoptes (python setup.py build install):

```
from pocs.state.machine import PanStateMachine

~/anaconda3/lib/python3.6/site-packages/pocs-0.6.0-py3.6.egg/pocs/state/machine.py in <module>()
      2 import yaml
      3 
----> 4 from transitions import State
      5 
      6 from pocs.utils import error

ModuleNotFoundError: No module named 'transitions'
```

What am I doing wrong ?
Thank you in advance for your help

AnthonyHorton, 2018-07-10:
Did you run the `install-dependencies.sh` script first (as mentioned in the README, https://github.com/panoptes/POCS/blob/develop/README.md)? Both `transitions` and `pymongo` are in the `requirements.txt` file so should have been installed by that script.

gnthibault, 2018-07-10:
Nop, my bad, sorry

closed 2018-07-10 by gnthibault

AnthonyHorton, 2018-07-10:
Not necessarily a non issue. That script is really for Debian based Linux (e.g. Ubuntu), which some users might not be using. We do have some instructions for OS X in the wiki though, which does include the `pip install -r requirements.txt` alternative: https://github.com/panoptes/POCS/wiki/Mac-OSX-Development-Install

wtgee, 2018-07-10:
Hi @gnthibault , thanks for jumping in!

@AnthonyHorton is right that you should run `pip install -r requirements.txt` first. This is a current mis-match between a pure `setup.py` install and a `pip install` that we want to resolve one day.

I'll address your other questions over in #533.