panoptes/POCS #536: install-dependencies.sh script fails on Ubuntu 18.04 LTS

AnthonyHorton, 2018-07-10:
The `install-dependencies.sh` script fails on Ubuntu 18.04 LTS with the following issues:

1. Attempt to `apt-get install` `libpng12-dev` (for `astrometry.net`) fails because `libpng12-dev` has been replaced by `libpng-dev` after Ubuntu 16.04 (https://askubuntu.com/questions/991706/e-package-libpng12-dev-has-no-installation-candidate). Replacing `libpng12-dev` with `libpng-dev` in `apt-packages-list.txt` fixes that, but would presumably break the script for old OS versions.

1. `apt-get` refuses to install MongoDB 3.2, because the `mongodb.org` repo for 18.04 doesn't have a release file:
```
Installing mongodb, for which several commands require sudo, so
you may be prompted for you password.

Executing: /tmp/apt-key-gpghome.YE6Dnn5Qd2/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv EA312927
gpg: key D68FA50FEA312927: public key "MongoDB 3.2 Release Signing Key <packaging@mongodb.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
deb http://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/3.2 multiverse
Hit:1 http://au.archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://au.archive.ubuntu.com/ubuntu bionic-updates InRelease      
Hit:3 http://au.archive.ubuntu.com/ubuntu bionic-backports InRelease    
Hit:4 http://au.archive.ubuntu.com/ubuntu bionic-security InRelease     
Ign:5 http://dl.google.com/linux/chrome/deb stable InRelease                        
Ign:6 http://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/3.2 InRelease           
Err:7 http://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/3.2 Release
  404  Not Found [IP: 54.192.134.137 80]
Hit:8 http://dl.google.com/linux/chrome/deb stable Release
Reading package lists... Done
E: The repository 'http://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/3.2 Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
Making this worse, once `install-dependencies.sh` has added the MongoDB 3.2 repo to the software sources list all subsequent `apt-get update`'s will error out until you manually remove/disable the repo.

With `libpng12-dev` replaced with `libpng-dev` and the `--no-mongodb` option `install-dependencies.sh` runs to completion with no errors (but no MongoDB, either).

AnthonyHorton, 2018-07-10:
This is a prerequiste for #531, given that the all-in-one install script plan has a nominal target of Ubuntu 18.04.

AnthonyHorton, 2018-07-10:
One more 18.04 `install-dependencies.sh` issue, the script adding `source activate panoptes-env` to my `.profile` broke my GDM login. Following a reboot I tried to login and got presented with a black screen with an error dialogue box stating:  

I couldn't dismiss the dialogue box and had to switch to a virtual terminal in order to comment out the line from my `.profile` and reboot again.  Before running `install-dependencies.sh` script I had already manually created a conda env for POCS and put a corresponding `source activate` line in my `bashrc`. Having it there doesn't seem to cause any problems.

jamessynge, 2018-07-11:
Thanks so much for all the details on the errors. I'll make use of it for the scripts.

Hopefully we'll be able to switch to Docker at some point, and maybe also away from Mongo which is the most difficult component to install.