panoptes/POCS #540: Separate target info from priority and other selection info

jamessynge, 2018-07-12:
I propose that the targets info should be available without any priority info, but with info that can be used to make decisions. For example, instead of having a priority and exposure time for a target, we could have a info about the magnitude of the star and, if known, the depth of transit(s). Ideally, we'd have access to transit ephemera, perhaps by querying a service.

We can then use separate file(s) to provide info about the mapping of magnitude to exposure, which can vary by scope. We can have magnitude limits for a site (e.g. PAN001 will be able to see a far dimmer star than PAN006 can). And we can have exposure limits for a site (PAN001 can take a longer exposure than PAN006). Finally, preference info such as how long to observe a target before and after the transit.

I'd like to use this sort of data to inform the selection process, so that I can use the same target info as any other unit, but provide different parameters for selecting targets. And it would be great to have a command line tool that allows me to get a "prediction" of what will be selected on the next few nights, assuming good weather.

I add issue this in the context of adding in targets for the next few nights where I need to edit the standard targets file (simple.yaml) in order to change priorities.

@wtgee: your thoughts?

wtgee, 2018-07-12:
Note a huge factor, but also note that scheduling on the Raspberry Pi takes a fairly significant time. Assuming even a modest web connection a query to a scheduler service would invariably be faster, especially as we add more constraints.

jamessynge, 2018-07-12:
Thanks for the detailed response. I agree that a web service is the way to go for the survey, and for priority overrides. We'll still need a local means to provide both extra constraints and priority boosts for locally selected targets. Since all units likely to be operational in the next 6 months are also likely to be connected to the internet, how about if we make it a priority to create the web service, taking as input the coordinates, obstructions, magnitude & exposure info, etc. This can start with the existing code.

I'd like to see it extended at some point to take into account the weather forecast so that we don't decide to prioritize observing a deep transit due at 1am, skipping a less good transit at 11, when the forecast is for rain starting at 12:30.

Is astroplan pure Python? And the rest of the dependencies for the scheduler? If so, could we trivially write this as a app engine app?