panoptes/POCS #556: Assess ability to use gphoto2's liveview with Canon cameras

jamessynge, 2018-08-25:
gphoto2 supports liveview, i.e. streaming over USB to gphoto2 the video that would normally be shown on the LCD on the back of the camera. This could potentially enable:

- Instantaneous feedback about focus changes.
- Periodic error correction using the DSLR instead of a guide camera.
- Displaying the live sky view on PAWS.

And I'm sure some other activities (suggestions welcome).

This issue is simply about experimenting with the live view feature and determining whether we can reasonably get access to the frames from Python.

jamessynge, 2018-08-25:
In interesting thing to consider is adding a microcontroller speaking the ST4 guiding camera interface to the mount. For example:

- https://hackaday.io/project/4386-arduino-st4-telescope-control
   - Github repo: https://github.com/kevinferrare/arduino-st4
   - Indilib driver: https://github.com/tliff/indi_ardust4
- https://github.com/Gordon999/Pi-AutoGuider
- https://github.com/Gordon999/Arduino-Uno-AutoGuider

This might be useful both for PEC training of the iOptron, but also for tracking adjustments during exposures.

wtgee, 2018-08-25:
Nice additions. I think we would want to just consider support a standard guider as many amateurs might have an extra one already.

Also, based off the PAN012 performance we almost don't need it. :grin:  Very curious to see how additional units perform.

kmeagle1515, 2019-03-22:
Hello sir!
Involving a rpi will lower the fps, but if the camera is directly connected with the monitor then more fps will be available. So what exactly you want to tryout?

wtgee, 2019-03-25:
> Hello sir!
> Involving a rpi will lower the fps, but if the camera is directly connected with the monitor then more fps will be available(comparatively). So what exactly you want to tryout?

Not sure if @jamessynge had thoughts on this, but for a guide scope the 25-30 fps should be okay. It would probably require a dedicated pi rather than using the pi that is also controlling the system.

This issue is mostly just a test idea, not something that we need or are looking at necessarily implementing right away.

wtgee, 2024-03-28:
This isn't well defined and a bit stale so I'm closing for now. There has been some conversation about using the liveview for focusing but a new specific issue should be created for that.

closed 2024-03-28 by wtgee