panoptes/POCS #573: Consider requiring Ubuntu 18.04 or later

jamessynge, 2018-08-31:
PAN006 is running Ubuntu 16.04.5, and the same may be true for PAN001.
We should consider whether (for the Intel NUC) we're ready to require Ubuntu 18.04, in part to make installations more consistent.

Alternately, we could more seriously explore Docker.

wtgee, 2018-09-01:
I definitely think we should do this. 

At some point early in PANOPTES we agreed that we would stay on top of updates to thinks like ubuntu, python, etc. This can of course be difficult with hardware and we obviously need to be careful about upgrading, but I'm definitely in favor of the ubuntu upgrade (and other LTS updates). We have few enough units right now that I think it's a good time.

AnthonyHorton, 2018-09-01:
I approve too. Some issues to address though: #536. Is a change of Mongo DB version needed?

wtgee, 2018-10-16:
I believe we've said yea? Closing.

closed 2018-10-16 by wtgee