panoptes/POCS #576: Reconsider supervisord

jamessynge, 2018-09-01:
#26 proposed using supervisord, but was closed. I wrote scripts (POCS/scripts/startup) for starting things inside a tmux session, but without the benefits of supervisord, i.e. no automatic restarting, no easy control such as stop all/selected and start all/selected. We should reconsider this "choice". Note that I'm not certain that supervisord is the perfect choice, but it is better than taking the time to write this feature ourselves.

Supervisord is appropriate for programs that do not have any command-line interaction, i.e. not our current shells, which do essentially all of the work of POCS; the exceptions are the messaging hub, social messaging script and PAWS, and for Huntsman the nameserver, etc. Those exceptions could be moved immediately to be run by supervisord, and overtime we can consider how to move more of the POCS system to run under supervisord.

@wtgee @AnthonyHorton Your opinions would be appreciated.

wtgee, 2018-09-01:
Adding @joshwalawender as the original suggestion to use supervisord was his as I believe he used it for a few projects so might have opinion.

Regarding the POCS shell, I think we (soon) should move to making the shell independent from the running process. Initially I was just going to `subprocess` the `pocs` instance (or something to that effect), but we could split it out entirely and control it via supervisord.

Sometimes I go back and forth on whether I do want automatic start-up after something like a power failure or if it should in some sense include a human sanity check on the unit first. But I suppose there is still a case for supervisord starting and running things once given that go-ahead.

wtgee, 2020-02-17:
Will be impacted by #951

wtgee, 2020-06-14:
Closing as stale.

closed 2020-06-14 by wtgee