panoptes/POCS #586: Fix sequence and image ids

wtgee, 2018-09-09:
The sequence_id and image_id (SEQ_ID and IMG_ID) are built from the `<unit_id>_<camera_id>_<sequence_time>` but the value for `unit_id` is being pulled from the `config['name']`. This has created problems for PAN006 where there is a custom name instead of a unit_id. Now that we actually have a unit_id parameter in the config, the sequence_id and image_id should be changed.

jamessynge, 2018-09-09:
Oh, I didn't realize there was an issue with my images. Do we need to run a
correction pass over the already uploaded observations?

wtgee, 2018-09-20:
> Oh, I didn't realize there was an issue with my images. Do we need to run a correction pass over the already uploaded observations?

We're going to have to do some scrubbing of a lot of files. Partly why I wanted to version bump (#166).

PR #613 (2018-09-21): Use unit_id for sequence and image ids
* **Behavior Change** Enables `pan_id` in the config by default. `PAN000` as simulator?

Closes #586
closed 2018-09-21 by wtgee