panoptes/POCS #587: Change *Dome.status to be a method

jamessynge, 2018-09-09:
As per https://github.com/panoptes/POCS/issues/189#issuecomment-354240479, status should be a method, not a property.

jamessynge, 2018-09-15:
@wtgee What should the return type be for status()? The property I added for the Astrohaven dome support is a string (e.g. "Open" or "Closed"). Most of the status() methods are returning dictionaries. Any thought on what should be done for the dome status?

wtgee, 2018-09-15:
I think we should do dictionaries so they are machine-readable in some sense. I think we will also have more than one status variable, e.g. battery level (if applicable, Huntsman has), parked status, connected, etc.

jamessynge, 2018-09-15:
In that case, it would help if we had some "standard" for what the various dictionaries should contain; for example, for status from a dome, should the dict contain the brand, model and driver (roughly a repeat of the data in the config)? That would allow the old config to be visible in logged status dicts (e.g. when examining the info saved along with the images of an observation).

Does huntsman have a standard for this already? dome/bisque.py has status as a property, but I'm not sure what the value type is.

jamessynge, 2018-09-15:
It looks like getting the value of Dome.status from dome/bisque.py will fail because it references a resource file `dome/status.py` that doesn't exist in the POCS repo.

@AnthonyHorton Do I have this correct?

wtgee, 2020-06-01:
Closed by #951. All hardware should have a status property now.

closed 2020-06-01 by wtgee