panoptes/POCS #606: Develop scheme for message topics

jamessynge, 2018-09-18:
We have a grab bag of existing topics used with PanMessaging. I'd like to develop a scheme for hierarchical topics, taking advantage of the prefix filtering that ZeroMQ offers. This should be recorded in a markdown file, probably under source control rather than in the wiki.

jamessynge, 2018-10-05:
Regarding topic names, given that ZeroMQ has literal prefix filtering (not regexp), I suggest that we use dotted names for topics (ala Python & Java package names). Furthermore, to eliminate the possibility of a leaf topic (i.e. the topic used by a publisher when sending a message) also matching as the prefix of some other, longer topic name, I suggest that all leaf topic names end with a `$` (dollar sign).
For consistency, I recommend that we only use lower-case names.

When a subscriber wants to filter on a group of topics with a common prefix, the filter should end with a `.` (period), unless subscribing to all topics, in which case the filter is the empty string. This reinforces the idea that dotted names are intended for semantic grouping. We could hide some of these details by introducing classes or class instances representing topics and topic groups. For example:
```
t = PanMessaging.create_topic_name('pocs_shell.in.command')
t  -->  TopicName('pocs_shell.in.command$')
g = t.parent_group_name()
g  -->  TopicGroupName('pocs_shell.in.')
g = g.parent_group_name()
g  -->  TopicGroupName('pocs_shell.')
g = g.parent_group_name()
g  -->  TopicGroupName('')
g = g.parent_group_name()
g  -->  TopicGroupName('')

g = PanMessage.root_topic_group_name()
g  -->  TopicGroup('')
```

We have these existing topics (excluding those used in tests):

- `PANCHAT` (sent from POCS.say)
  This is used to broadcast messages intended for humans. Subscribed to by PAWS and social_messaging.
  Suggested name: `pocs.out.chat$`
- `PAWS-CMD` (sent from PAWS websockets.py, received by pocs_shell)
  Suggested name: `pocs_shell.in.command$`
  This should probably be combined with `POCS-CMD` once we create a non-shell version of pocs_shell (i.e. a daemon that runs the observatory, receives commands, reports status, etc.).
- `POCS` (sent by class POCS)
  This is the default topic for POCS.send_message, but I think all the non-test callers provide a topic, and only one test does not provide a specific topic. We should probably eliminate this topic.
- `POCS-CMD` (sent by pocs_shell, received by class POCS, and also by some tests in order to shutdown POCS)
  Suggested name: `pocs.in.command$`
- `STATUS` (sent by class POCS)
  I'm not entirely clear how this is received, but I assume that it some how goes from POCS to PAWS to the browser javascript.
  Suggested name: `pocs.out.status$`
- `weather` (from peas/sensors.py)
  Suggested name: `sensor.weather$`
- `environment` (from peas/sensors.py)
  I suggest that we drop this in place of the following two messages which ArduinoIO will produce (the names are derived automatically from the board name).sensor.weather$
- `telemetry:commands` (in ArduinoIO... new)
  Suggested name: `sensor.telemetry_board$`
- `camera:commands` (in ArduinoIO... new)
  Suggested name: `sensor.camera_board$`

wtgee, 2020-06-01:
Messaging was eliminated in #951. The ideas here are relevant but it might also be worth just starting a new issue. Closing as stale.

closed 2020-06-01 by wtgee