panoptes/POCS #629: dispatch scheduler question

gnthibault, 2018-09-25:
Dear all,

I am still playing with POCS, and I recently came across a line of code that I suspect might generate a bug.
However, I am sorry, I don't have enough time to write proper to test to make it fail consistently.

Here: https://github.com/panoptes/POCS/blob/develop/pocs/scheduler/dispatch.py#L92

I think you should be inserting a tuple instead of just and observation:

`best_obs.insert(0, (self.current_observation.name, self.current_observation.merit))
`
Sorry if I am wrong

I wish you a good day

wtgee, 2018-09-29:
Hi @gnthibault, thanks for pointing this out! You are right that it should be a tuple that is being inserted and i have made a fix to make it consistent. This was not being used for operating our units but potentially could be by someone, so I appreciate it.

Cheers!

gnthibault, 2018-09-29:
You are most welcome! I modified the dispatch class in order to use it, and I might have some other correction to report.

If you have a bit of time, I have two questions regarding pocs:

- What version of astroplan are you using (I ambusing the latest one, and it does not look fully compatible with pocs)
- Are you actually uing astroplan scheduling capability in your own scheduler implementation (I am curious)

I wanted to use astroplan scheduling capability, but it looks like this dispatch class is enough for my constraints, and what I need to do right now.

Thank you.

closed 2018-09-30 by jamessynge

jamessynge, 2018-09-30:
Thanks again for the fix. We use the latest version of astroplan (i.e.
requirements.txt just says "astroplan" with no version constraint). @wtgee
can address the question about whether/how we are using astroplan.

PR #641 (2018-09-30): Make sure the merit gets added to the best observations list.
Fix an inconsistency in assigning to the list of best observations. Because `self.current_observation` is assigned as well (which has a `.merit` property), this list is not actually used for anything during operations and we don't even capture the return value. But should be consistent regardless.

Closes #629.