panoptes/POCS #631: exp_time vs exptime consistency

wtgee, 2018-09-25:
We currently use both `exp_time` and `exptime` as a variable name in different places. For clarity we should pick one and always use it.

Unless there is a good reason otherwise let's start using `exptime`.

AnthonyHorton, 2018-09-26:
`EXPTIME` is the most common form for FITS headers so using `exptime` in our code makes sense.  On the other hand `exp_time` is more in keeping with Python conventions for variable names. 3rd option, replacing them all with `exposure` instead! As its one word there's no question about whether it should have underscores or not.  `EXPOSURE` is also an accepted FITS header keyword, albeit in less common usage.

ImmortalRabbit, 2019-03-04:
> `EXPTIME` is the most common form for FITS headers so using `exptime` in our code makes sense. On the other hand `exp_time` is more in keeping with Python conventions for variable names. 3rd option, [...]

Can I try to solve this issue? I am planning to replace EXPTIME to EXPOSUE and exptime, exp_time to exposure. Do you think that it will solve this issue?

commit 3216e342 (2019-03-15): Changing exptime variable names to be consistent

Thanks for the work @ImmortalRabbit!

Closes #631
closed 2019-03-15 by wtgee