panoptes/POCS #632: Figure out why test_autofocus_coarse hangs

wtgee, 2018-09-25:
The `test_observatory.py` `test_autofocus_coarse` is hanging on almost all branches some of the time. Often a restart of the test will fix it. Need to figure out what process is causing the problem.

wtgee, 2018-09-25:
Setting `plots=False` in the test files seems to do the trick. Still testing out some changes.

wtgee, 2018-09-26:
Note: I didn't figure out *why* this was happening, I just set it to skip for certain tests. Still need to figure out why so should probably leave open.

AnthonyHorton, 2018-09-26:
That is odd. Haven't had issues with this in the past. I have been seeing intermittent test failures in the Arduino tests, though.

This change to disable focus plots in the tests has broken the autofocus tests in my `pyro-cameras` branch as there I had expanded the tests to check for the plot files while I was working on the copying of files from remote cameras. I've re-enabled plotting in my branch for now, will keep an eye out for hanging tests.

wtgee, 2018-09-26:
> That is odd. Haven't had issues with this in the past. I have been seeing intermittent test failures in the Arduino tests, though.

Something about the arduino simulator definitely does wacky things to our coverage.

> This change to disable focus plots in the tests has broken the autofocus tests in my `pyro-cameras` branch as there I had expanded the tests to check for the presence of the plot files while I was working on the copying of files from remote cameras. I've re-enabled plotting in my branch for now, will keep an eye out for hanging tests.

It definitely could have been the result of my refactor and/or subsequent barrage of PRs.

PR #633 (2018-09-26): Autofocus Plots set to False
* Setting autofocus plots to False by default.
* Pass autofocus params straight through to underlying `focuser.autofocus`.

Closes #632
PR #696 (2018-10-18): Fix autofocus tests
The coarse focus no longer also does a fine focus (#659)

Closes #677 and closes #632
closed 2018-10-18 by jamessynge