panoptes/POCS #653: Arduino Simulator Fixes

wtgee, 2018-10-01:
I am getting errors in the arduino simulator tests in unrelated PRs. In addition to that. the `pocs/serial_handlers/protocol_arduinosimulator.py` file reports coverage tests on almost every PR, pointing to some non-hermetic testing.

jamessynge, 2018-10-05:
I'm not able to reproduce this problem (I have in the past though, so not denying the problem still exists). Can you provide steps?

wtgee, 2018-10-05:
> I'm not able to reproduce this problem (I have in the past though, so not denying the problem still exists). Can you provide steps?

I cannot rely reproduce. But I think this is why we get that fluctuation 1.55% diff coverage from protocol_arduinosimulator. Some times I think it passes and sometimes not, without causing tests to fail (which I thought I had commented on elsewhere as well but don't see).

jamessynge, 2018-10-16:
I'm going to claim this is fixed by the large number of changes to the arduinio io code that I merged over the last few days. Let me know if that's wrong.

closed 2018-10-16 by jamessynge