panoptes/POCS #656: Decouple camera from observation

wtgee, 2018-10-01:
The `take_observation` command has some coupling between the camera and the observation that should be removed. See discussion here: https://github.com/panoptes/POCS/pull/642#discussion_r221471935

AnthonyHorton, 2018-10-02:
Going to take a little bit of thought to neatly disentangle the camera and observation because of the desire to insert of bunch of observation related metadata into the FITS headers of the images the camera is taking.

jamessynge, 2018-10-19:
Is there anything that prevents us from updating the images after they've been collected by the camera? Alternately, the camera object could be provided either with a set of metadata to add to the FITS headers or with a callback for doing the same (the latter is probably much more general).