panoptes/POCS #662: Exception collecting status

jamessynge, 2018-10-02:
Executed after updating POCS to commit 85c3a3a2448ef4ad8fded07f4000370256b65d07:

```
$ bin/pocs_shell
POCS > setup_pocs
POCS > collect_status
Starting messaging
Command publisher started on port 6500
Command subscriber started on port 6501
Message subscriber started on port 6511
Message publisher started on port 6510
Recording POCS status - Press Ctrl-c to interrupt
{'observatory': {'dome': 'Both sides closed',
                 'mount': {'current_dec': <Latitude -67.67124444444445 deg>,
                           'current_ha': <Longitude 23.999750804119877 hourangle>,
                           'current_ra': <Longitude 157.70974999999999 deg>,
                           'dec_guide_rate': 0.9,
                           'gps': 'Data Extracted',
                           'hemisphere': 'Northern',
                           'movement_speed': 'Max sidereal',
                           'ra_guide_rate': 0.9,
                           'state': 'Stopped - Not at Zero Position',
                           'time_source': 'RS-232',
                           'timestamp': '-3000181002093034',
                           'tracking': 'Custom',
                           'tracking_rate': '1.0000',
                           'tracking_rate_ra': 1.0},
                 'observer': {'local_evening_astro_time': <Time object: scale='utc' format='jd' value=2458394.498133679>,
                              'local_moon_alt': <Latitude 38.92783285076994 deg>,
                              'local_moon_illumination': 0.47878615636559624,
                              'local_moon_phase': <Quantity 1.6132367534603926 rad>,
                              'local_morning_astro_time': <Time object: scale='utc' format='jd' value=2458394.8826049455>,
                              'local_sun_rise_time': <Time object: scale='utc' format='jd' value=2458393.9494617917>,
                              'local_sun_set_time': <Time object: scale='utc' format='jd' value=2458394.4306360562>,
                              'localtime': '2018-10-02 10:30:33',
                              'siderealtime': '10h30m54.6383s',
                              'utctime': <Time object: scale='utc' format='datetime' value=2018-10-02 14:30:33.291914>}},
 'state': 'sleeping',
 'system': {'free_space': 137.299013632}}
********************************************************************************
Traceback (most recent call last):
  File "bin/pocs_shell", line 904, in <module>
    PocsShell().cmdloop()
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/cmd.py", line 138, in cmdloop
    stop = self.onecmd(line)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/cmd.py", line 217, in onecmd
    return func(arg)
  File "bin/pocs_shell", line 228, in do_collect_status
    self.pocs.db.insert_current('system', status, store_permanently=False)
  File "/var/panoptes/POCS/pocs/utils/database.py", line 253, in insert_current
    self.validate_collection(collection)
  File "/var/panoptes/POCS/pocs/utils/database.py", line 41, in validate_collection
    raise InvalidCollection(msg)
pocs.utils.error.InvalidCollection: Collection type 'system' not available
```

wtgee, 2018-10-02:
Hmm, that's an odd new one. Fix is easy but it concerns me why that has never happened but suddenly is.

wtgee, 2018-10-02:
I see that this is via `collect_status`, which shouldn't matter, but does this happen during operations?

wtgee, 2018-10-02:
I've confirmed I get the same thing on PAN001 from https://github.com/panoptes/POCS/commit/77cd582df1adaaf5a79f53fe7473ec761f189192. Normal operations have been okay.

Honestly I forgot there was a `collect_status` call in pocs_shell. That sure is handy. :grinning:. Let's mark as bug and I'll try to dig into the shell but I think you should be okay for running.

wtgee, 2018-10-02:
Looking into this, the `collect_status` is just reading the status and then trying to insert it into a whole new collection named `system`, which we don't use anywhere else. Since the status that is returned is really just items that are already stored in other collections I don't know what the point would be of restoring it. 

I think the answer here is to just remote the line that [inserts this](https://github.com/panoptes/POCS/blob/develop/bin/pocs_shell#L227).