panoptes/POCS #697: Add AC sensor signal into safety decision of POCS

jamessynge, 2018-10-17:
@wtgee Per our discussion with Nem today.

jamessynge, 2018-10-19:
Specifically, POCS.is_safe should add a new entry to the is_safe_values dictionary for AC. The environment data for the telemetry board already has the info we're looking for; it _simply_ needs to be extracted and compared against a suitable threshold. Unfortunately, that threshold is likely to be unit specific.

FWIW, I'd really like to change the design so that we get a binary signal indicating if AC is present or not, without a comparison required. Using the Mean Well brand power supply would provide that. Similarly, just plugging in a 5V wall wart (e.g. the type used to power an Arduino UNO) could be used to close or open a relay, and we could use a digital IO pin on the Arduino to detect that the AC to 5 VC DC conversion is happening. It isn't quite the same as having the power supply tell us that it has 5V, but it is easy to create out of standard parts.

wtgee, 2018-10-19:
> FWIW, I'd really like to change the design so that we get a binary signal indicating if AC is present or not, without a comparison required

I'd like this. It would be great if we could just read a boolean flag out of the arduino.

I'll get to this work this week. Thanks for writing up Issue.