panoptes/POCS #698: Rename scheduler target specifciation keys

wtgee, 2018-10-18:
The terminology to specify targets is confusing.  Specifically, `min_nexp` should be renamed to `min_exp_sets` and some examples of how to specify targets should be given.

ImmortalRabbit, 2019-03-19:
```
        Keyword Arguments:
            exptime {u.second} -- Exposure time for individual exposures
                (default: {120 * u.second})
            min_nexp {int} -- The minimum number of exposures to be taken for a
                given field (default: 60)
            exp_set_size {int} -- Number of exposures to take per set
                (default: {10})
            priority {int} -- Overall priority for field, with 1.0 being highest
                (default: {100})

That's my changes, what do you think, sorry I am not talented to chose right variables but I can change them to what ever variable you ask me to change.

min_nexp change to chose_best_one(min_exp_set, min_num_exp, min_exps, 
exp_set_size change to chose_best_one(exps_per_set, num_exp_per_set, num_exp_per_set)
priority change to chose_best_one(field_priority, prioriy_field, f_priority)

```

wtgee, 2022-01-20:
This is stale and apparently not necessary. Reopen if needed.

closed 2022-01-20 by wtgee