panoptes/POCS #707: Calibrate zero (home) position automatically of iOptron mount

jamessynge, 2018-10-20:
The iOptron iEQ 30 Pro or 45 Pro mounts don't have absolute encoders on the RA and DEC shafts, so they can only make relative motions. This means that they need to be moved "manually" into a position called the zero or home position where the scope is pointing at the pole and the DEC axis is as vertical as it can be. The user then needs to use the iOptron hand-controller to set the current position as the zero position. After that, the mount can move to specific coordinates by making movements relative to that zero position.

If you unlock the clutches on the iOptron iEQ 30 Pro or 45 Pro, adjust the position of the mount and then lock the clutches, the zero position will not be what it was prior to that adjustment, so you need to set the zero position again. However, this is really hard to do during the day, the time when it is easiest to work on the mount (or show it off to visitors). I think we can assume that the mount will be left in approximately the correct parked position for safety, but that might not be exactly the same as the previous parked position, so the relative travel from parked to zero will have changed.

Currently POCS assumes that the zero position is set correctly, but for at least PAN006 this isn't correct. We could use several means to validate the mount location when starting up:

- The accelerometer could be checked to see if the mount has moved since it was parked (i.e. is down still in the same direction it was after parking).
- After moving to the zero position, POCS could take a picture of the pole (or of several locations) and solve the image(s) to determine the actual orientation of the mount.

After all that, we can move the mount to the position that should be zero and set that position as zero. We might want to do that two or three times to get as close as possible.

Requesting the feedback of @wtgee and @joshwalawender.

wtgee, 2018-10-20:
>  The user then needs to use the iOptron hand-controller to set the current position as the zero position.

Actually we can set this without the hand-controller, which was what I was referring to in our offline conversation. There is a private method [`_set_zero_position`](https://github.com/panoptes/POCS/blob/develop/pocs/mount/ioptron.py#L332) which calls the underlying command to set the zero position.

> After moving to the zero position, POCS could take a picture of the pole (or of several locations) and solve the image(s) to determine the actual orientation of the mount.

As mentioned, the plate-solving near the pole can be difficult to get fairly accurate. The accelerometer is a good idea.

I don't think normal operations of the mount (outside of PANOPTES) would rely on any super accurate home position, but instead use the calibration obtained via an alignment procedure to correct for the discrepancies. And, even if the home position was 100% "accurate" I think you would still need the alignment calibration.

That alignment calibration is done via the `calibrate_mount` (see below) command, which we currently use in the pointing correction, but that was some of the info we were hoping to get back from iOptron. 

While I think recovering the home position is a good idea I think our time is probably better spent figuring out the details of the `calibrate_mount` command and how to "properly" do an alignment via software only.

I also do like the idea of figuring out if the unit has moved while parked, e.g. via the accelerometer.

Note: At this point `calibrate_mount` we don't have a high-level wrapper for this command although you can see usage via `mount.query` in the [pointing state](https://github.com/panoptes/POCS/blob/develop/pocs/state/states/default/pointing.py#L88) when it does the auto-correct.

wtgee, 2022-01-20:
The newer versions of the iOptron mounts do have some kind of absolute encoder and have the ability to search out the home position automatically. Closing this issue as Won't Fix unless we get strong motivation to support, in which case Issue can be re-opened.

closed 2022-01-20 by wtgee