panoptes/POCS #711: Effectively cache astrometry, et al, on travis

jamessynge, 2018-10-22:


wtgee, 2018-10-22:
I have a branch where I started some work on this. I'll see if it's worth pushing.

jamessynge, 2018-10-23:
Assigning it to you in hopes that your branch has something worthwhile in it. Several recent builds have failed due these problems.

jamessynge, 2018-10-23:
Alternately, point me to your branch.

jamessynge, 2018-10-27:
I've made lots of progress on understanding how to set this up (#712), but now am stymied by astrometry.net being offline again. It seems like we should cache critical dependencies in GCS.

FWIW, it appears that I spoiled the cache for #712 with one bad command which then meant that the next build of that PR didn't have a cache to pull from, so needed to read from astrometry.net. I'm going to try to get that PR thing submitted today so that future builds will hopefully be better able to use the travis provided caching facility.

wtgee, 2019-03-29:
@jamessynge did we ever get anywhere with this?

If anything I am thinking that as we move more toward a docker-based approach it won't be an issue. I know we are also storing a few things in our own storage bucket, which it doesn't appear we are using but which we could.

wtgee, 2020-06-01:
Made stale by #951. Please re-open a new issue if needed. Thanks!

closed 2020-06-01 by wtgee