panoptes/POCS #719: Remove PANUSER environment variable?

jamessynge, 2018-11-03:
This is only used by scripts during the creation of /var/panoptes, during which we want to set the ownership of the directory, but not at any other time. We do use PANUSER in some documentation to guide the user.

We should consider removing it.

ImmortalRabbit, 2019-03-10:
> Not opposed to removing it if we don't appear to be using it.
If I will decide to fix this issue, will it be enough to properly delete PANUSER variable from POCS and replace with panoptes? Is there anything that in travis-CI also should be fixed or just to change POCS will be enough to this issue?

AnthonyHorton, 2019-03-11:
@ImmortalRabbit It will take a bit more than just replacing the `PANUSER` environment variable with `panoptes` everywhere. As @wtgee notes there are sometimes cases where user isn't `panoptes`, e.g. in `huntsman-pocs` and, if I recall correctly, when running in Travis CI.  What we would need is an alternative way of setting `PANUSER`, probably in the main config file.

wtgee, 2020-02-16:
I'm not sure this is realistic or desirable at this point. Closing as stale.

Related to #555

closed 2020-02-16 by wtgee