panoptes/POCS #743: Backup cloud scheduler

wtgee, 2018-11-29:
As a first step toward having a networked scheduler it would be nice to provide a simple cloud function that can return available "pretty image" fields.

This is motivated by the fact that I want PAN001 to dwell on the TESS sectors as specified in #742. However there are inevitably going to be some times where the unit cannot schedule a full two hour observation block. I don't want to pollute this file with other targets so it would be nice if during these scheduling attempts we could query a cloud function that could return a field that could be observed for 20-30 minutes.

Note that it is not at all necessary to do this with a cloud function or via the network but the idea would be to use it as a stepping stone toward a cloud-based scheduler.

Thoughts? @jamessynge @joshwalawender

jamessynge, 2018-11-30:
Go for it!

joshwalawender, 2018-12-04:
Sounds good.  We should carefully think through the implications of going 100% cloud scheduled for the survey before going too far down this path as we had originally planned to allow units to be unconnected to internet.  We might drop that and require a connection, but we should explicitly make a decision on whether or not we do that. This sounds like a reasonable way to explore that avenue without committing us.

jamessynge, 2018-12-04:
I think we need the option of completely disconnected operation to remain
viable, especially for survey mode. But we should also allow for many, if
not most, scopes to have a good internet connection that we can take
advantage of, both for coordinated campaigns and to enable the option of
skynet style scheduling.

wtgee, 2018-12-04:
The idea would definitely be to have this as an item in the config that can easily be turned on and off. I don't think we would ever really go 100% cloud scheduled, at least not in the sense that all units are required to query a central server before beginning nightly operations. There will inevitably (hopefully) be units that are remote and isolated (off-grid) that might only very infrequently want to connect to the internet.

We also need to keep in mind that there could be complications with internet connections for any units that are hosted at public schools as there will inevitably be firewall rules and protections in place.

I think we are all on the same page on this, i.e. supporting the ability to do cloud-based scheduling but not relying on it.

wtgee, 2018-12-07:
Just as another small point although this is mostly the same as the pretty image. I kind of want to try to capture the comet 46P/Wirtanen. Obviously having a changing RA/Dec for a target would just be annoying for each unit to have to do whereas if we manage it on the server the unit doesn't need to care about that. Mostly putting this here for my own future reference.