panoptes/POCS #764: Consider making timelapse from fits

wtgee, 2018-12-11:
I mostly give the timelapse a quick look in the morning to see if there is anything unusual in the shots. In taking a [long series of 46P-Wirtanen](http://www.panoptes-data.net/observations/PAN001_14d3bd_20181211T064529) last night it is obvious that the comet if fairly hard to see and if I didn't know it was there I would have most likely missed it. In fact, to find where it was in the frame I had to look at the scaled image from the FITS file.

We should consider doing a diff-frame timelapse or just using some scaled images, which would make changes from one frame to the next more obvious.

AnthonyHorton, 2018-12-12:
Oh, I like the idea of a difference timelapse.

Abhinay1997, 2019-04-15:
@wtgee @AnthonyHorton, I'm working on a timelapse tool using FITS images.
Could you give some pointers on aligning PANOPTES images using WCS header info ? 
<b>Consider:</b>
I have two images: fits_1 and fits_2. They are successive images pointing at the same region in the sky.
I create a WCS object using astropy.wcs.WCS for both fits_1 and fits_2. 
So we have wcs_1 and wcs_2 corresponding to the two images.
<b>What I am trying to do:</b>
fits_1 chosen reference pixels --> RA and DEC values using wcs1.all_pix2world
RA and DEC values of reference pixels --> fits_2 pixel coordinates ( using wcs2.all_world2pix)
The problem is the shift between the pixels with same RA and DEC values in fits_1 and fits_2 is as follows:
```
x shift is -0.06906008065834612 pixels and y shift is 3.117477370702545 pixels

x shift is -0.06914311006084972 pixels and y shift is 3.1175005897762276 pixels

x shift is -0.07021917960992141 pixels and y shift is 3.1174266885186626 pixels

x shift is -0.06887403351583998 pixels and y shift is 3.1175184556773274 pixels 
```
<b>P.S</b>
The RA-MNT and DEC-MNT are same for both fits_1 and fits_2 but HA-MNT is different.
Would it be possible to align using this ?
Right now I'm trying to find a homography between fits_1 and fits_2 as a means to align but wanted to know if I'm moving in the right direction. Thank you in advance :) 
[jupyter notebook for reference](https://github.com/Abhinay1997/DIFFITS-Timelapse/blob/master/fits_align_with_wcs.ipynb)

Abhinay1997, 2019-04-16:
Hello @wtgee @AnthonyHorton . I am trying to build this tool and I am having trouble aligning the images.  Identifying the shift in pixels from reference image to the next image {using astropy wcs all_pix2world and all_world2pix} shows different shifts for different pixels. 
I then tried using astropy reproject package but apparently it doesn't utilize SIP information in the headers and even then it is quite slow.
Right now, I'm trying to figure out how SAO ds9 is able to match mutliple images using wcs near instantaneously.
Could you point me towards a resource that describes aligning images with wcs ?
[jupyter notebook for reference](https://github.com/Abhinay1997/DIFFITS-Timelapse/blob/master/fits_align_with_wcs.ipynb)

wtgee, 2019-08-25:
See https://github.com/panoptes/panoptes-timegen

wtgee, 2020-02-13:
Closing, see above comment.

closed 2020-02-13 by wtgee