panoptes/POCS #792: Default Ubuntu setup on 18.04 conflicts with using Chrome Remote Desktop

jamessynge, 2019-03-09:


wtgee, 2019-03-09:
What about just using [Xubuntu](https://xubuntu.org/) or [Lubuntu](https://lubuntu.net/)? They are both lightweight and nice. I think PAN012 is running Xubuntu. Then we are not only still Debian but we are still actually on [official Ubuntu](https://www.ubuntu.com/download/flavours). They still ship with e.g. office applications but would still be considerably lighter.

AnthonyHorton, 2019-03-11:
Does LightDM have better compatibility than gdm3? While Ubuntu now defaults to using GDM it continues to include LightDM by default so it's easy for the user to switch. Might be an acceptable short term fix.

jamessynge, 2019-03-16:
@wtgee I'd be happy to try a lighter weight install. I was wondering about using a distribution optimized for running containers or kubernetes, with the idea of pushing towards remote mgmt, but I fear that would be too big a hurdle any time soon. I'll give one of those two (Xubuntu and Lubuntu) a try today.

@AnthonyHorton wrote 

> Does LightDM have better compatibility than gdm3?

Better compatibility with what?

AnthonyHorton, 2019-03-17:
> > Does LightDM have better compatibility than gdm3?
> 
> Better compatibility with what?

With Chrome Remote Desktop.

programatt, 2019-03-29:
@jamessynge I have had very good success running a docker environment on stock ubuntu server instances. If people are ok with administration of the devices being on the CLI and eventually going to updates through docker images there's a really nice tool called [watchtower](https://github.com/v2tec/watchtower) that I use. It listens to the docker socket on the host and automatically pulls down new docker container images and restarts services when changes are detected. Can run on a schedule or every X seconds. For os level updates then you can use tools like ansible to run sh scripts against remote hosts. But it really depends on what the project goals are for the admin side.

wtgee, 2019-06-22:
Should be addressed by #778

wtgee, 2020-02-17:
Closing as stale.

closed 2020-02-17 by wtgee