panoptes/POCS #793: ImportError while loading conftest

ImmortalRabbit, 2019-03-10:
I trying to install POCS using [Coding in PANOPTES]( https://github.com/panoptes/POCS/wiki/Coding-in-PANOPTES). 
OS: ubuntu 16.04
I have cloned POCS into` /var/panoptes/`
I have set all environmental variables, installed mongodb, git, ghoto2, astronomy.net, dcraw and exiftool.
Successfully installed requirements using: `pip3 install -r requirements.txt` 
I have opened POCS using pycharm, and then I run `pytest` which gives me ERROR:

```
ImportError while loading conftest '/var/panoptes/POCS/conftest.py'.
conftest.py:16: in <module>
    from pocs.utils.database import PanDB
E     File "/var/panoptes/POCS/pocs/utils/database.py", line 143
E       raise ConnectionError(f'Mongo server not available')
E                                                         ^
E   SyntaxError: invalid syntax
```

I cannot understand what I have left to do or how to fix this error. 

I am trying to contribute with less git pushes because each time I have to git push and do pull requests in order to check all tests using travis-CI.

ImmortalRabbit, 2019-03-10:
That's was error with syntax because virtual environment was using not python3.6 which is required. However, in documentation it was said that:
> We recommend a newer python (3.5) separate from whatever system python is already installed. 
I think it confuses newcomers like because POCS uses some a new features of python3.6 which is not supported in python3.5. For example
` raise ConnectionError(f'Mongo server not available')`
In this line of code `f` is not supported by python3.5.  
> PEP 498 introduces a new kind of string literals: [f-strings](https://docs.python.org/3/whatsnew/3.6.html), or formatted string literals.

ImmortalRabbit, 2019-03-10:
Currently with python3.6.7, I am having error: 
```
/var/panoptes/POCS/pocs/utils/logger.py:218: UserWarning: formatters is missing from log_config!
  warn('formatters is missing from log_config!')
/var/panoptes/POCS/pocs/utils/logger.py:219: UserWarning: log_config: {}
  warn(f'log_config: {log_config!r}')
ImportError while loading conftest '/var/panoptes/POCS/conftest.py'.
conftest.py:18: in <module>
    from pocs.utils.messaging import PanMessaging
pocs/utils/messaging.py:18: in <module>
    class PanMessaging(object):
pocs/utils/messaging.py:80: in PanMessaging
    logger = get_root_logger()
pocs/utils/logger.py:262: in get_root_logger
    logging.config.dictConfig(log_config)
/usr/lib/python3.6/logging/config.py:802: in dictConfig
    dictConfigClass(config).configure()
/usr/lib/python3.6/logging/config.py:498: in configure
    raise ValueError("dictionary doesn't specify a version")
E   ValueError: dictionary doesn't specify a version

```

ImmortalRabbit, 2019-03-14:
> Currently with python3.6.7, I am having error:
> 
> ```
> /var/panoptes/POCS/pocs/utils/logger.py:218: UserWarning: formatters is missing from log_config!
>   warn('formatters is missing from log_config!')
> /var/panoptes/POCS/pocs/utils/logger.py:219: UserWarning: log_config: {}
>   warn(f'log_config: {log_config!r}')
> ImportError while loading conftest '/var/panoptes/POCS/conftest.py'.
> conftest.py:18: in <module>
>     from pocs.utils.messaging import PanMessaging
> pocs/utils/messaging.py:18: in <module>
>     class PanMessaging(object):
> pocs/utils/messaging.py:80: in PanMessaging
>     logger = get_root_logger()
> pocs/utils/logger.py:262: in get_root_logger
>     logging.config.dictConfig(log_config)
> /usr/lib/python3.6/logging/config.py:802: in dictConfig
>     dictConfigClass(config).configure()
> /usr/lib/python3.6/logging/config.py:498: in configure
>     raise ValueError("dictionary doesn't specify a version")
> E   ValueError: dictionary doesn't specify a version
> ```
> Can someone give any advice about fixing this issue?

I could fix it by properly adding environmental variables.

closed 2019-03-14 by ImmortalRabbit

wtgee, 2019-03-14:
> I could fix it by properly adding environmental variables.

Hi @ImmortalRabbit, sorry for the delay, I've been out for a few days.

Can you be specific about what environment variable you added, just in case someone else comes looking. Thanks!

ImmortalRabbit, 2019-03-19:
> Can you be specific about what environment variable you added, just in case someone else comes looking. Thanks!

I am not sure why but each time environmental variables reset's itself after restart of my PC, and each time I have to install all environmental variables again. I mean doing this again. 

export PANDIR=/var/panoptes   # Main Dir
export PANLOG=${PANDIR}/logs  # Log files
export POCS=${PANDIR}/POCS    # Observatory Control
export PAWS=${PANDIR}/PAWS    # Web Interface
export PIAA=${PANDIR}/PIAA    # Image Analysis
export PANUSER=panoptes       # PANOPTES linux user

wtgee, 2019-03-19:
You can put them in your `$HOME/.bashrc` file and then they should be loaded every time.