panoptes/POCS #795: Update minimum required Python Version

wtgee, 2019-03-10:
We are using f-strings from python 3.6 but the docs say python 3.5 is required. Need to get docs updated. Thanks @immortalrabbit

That's was error with syntax because virtual environment was using not python3.6 which is required. However, in documentation it was said that:
> We recommend a newer python (3.5) separate from whatever system python is already installed. 

I think it confuses newcomers like because POCS uses some new features of python3.6 which is not supported in python3.5. For example
` raise ConnectionError(f'Mongo server not available')`
In this line of code `f` is not supported by python3.5.  
> PEP 498 introduces a new kind of string literals: [f-strings](https://docs.python.org/3/whatsnew/3.6.html), or formatted string literals.

Should I create a new issue about this problem and fix it? @wtgee

_Originally posted by @ImmortalRabbit in https://github.com/panoptes/POCS/issues/793#issuecomment-471278269_

ImmortalRabbit, 2019-03-11:
@wtgee I am kind of lost with this issue because it seems like there is no way of doing poll request for wiki page. So what should I do to make you to review my changes? Here is my changes: [link](https://github.com/ImmortalRabbit/POCS/tree/wiki)

wtgee, 2019-03-15:
I have this changed [here](https://github.com/panoptes/POCS/wiki/Coding-in-PANOPTES#python). Anaconda is 3.7 be default now so I used slightly different wording than what you had.

Thanks for pointing this out!

closed 2019-03-15 by wtgee