panoptes/POCS #800: GSoC proposal

ImmortalRabbit, 2019-03-20:
Some of the organizations have a proposal format or template according to which applicants have to make the GSoC proposal. Is there any specific format that we have to follow with project Panoptes? Or is there any requirements or nice to haves for proposal?

@wtgee @jamessynge @mimming @joshwalawender @AnthonyHorton

wtgee, 2019-03-20:
Good question! :wink: We'll figure it out and get back to you shortly.

kmeagle1515, 2019-03-20:
yeah! have the same doubt.

ImmortalRabbit, 2019-03-25:
@wtgee Sorry for asking again but will we have answer soon? Or there will be no template or requirements for GSoC proposal? I am asking you again because  we have to start sending our proposals from tomorrow. 

Thank you for any response,
Demezhan Marikov.

wtgee, 2019-03-25:
Sorry @ImmortalRabbit , I had posted this [over on the forum](https://forum.projectpanoptes.org/t/proposal-making/243/4) but forgot to post here!

https://github.com/panoptes/POCS/wiki/GSoC-2019-Proposal-Template

Thanks for the prompt @ImmortalRabbit . I'm going to close the issue.

closed 2019-03-25 by wtgee

ImmortalRabbit, 2019-03-25:
I decided to write my questions about GSoC ideas here, rather than creating a new issue. If I understood correctly point of doing suggested milestones is helpful to understand the project idea and to get start doing basics. If we have already finished it and would like to know is it correct start or not, can we do pull request and ask mentors to check our code? @wtgee @jamessynge @mimming @joshwalawender @AnthonyHorton

wtgee, 2019-03-26:
Hi @ImmortalRabbit, not sure I understand the question. If you have already started on one of the ideas and have some PRs then yes, please submit them. It will help us to define the scope of the projects or what work can actually get done over the course of the summer.