panoptes/POCS #803: How to test removed items from observatory

ImmortalRabbit, 2019-03-27:
## Expected Behavior

There should test methods that checks properly methods that removes camera, dome and mount.

## Current Behavior

Currently, I use for a test: 
`  @property
    def exist_dome(self):
        return 'dome' not in locals()`
It just checks is variable dome exist in local variables or not. Test goes very well, but still I am not sure of correctness of this method. 
## Possible Solution

` try:
      dome
     return True
  except(Expection ex):
      return False
`
Also, I think it could be possible solution. 

## Context

I am trying to implement suggested milestones of "Observatory Dependency Decoupling". I am currently on a third part of it where I am trying to implement tests for methods that removes camera, mount and dome. 

## Your Environment

* Version used: python3.6.7
* Environment name and version (e.g. Python 3.6.7 on conda 4.6.3): python3.6.7 on virtualenv 
* Server type and version: localhost
* Operating System and version: Ubuntu16.04
* Link to your project: https://github.com/ImmortalRabbit/POCS/tree/gsoc_test

AnthonyHorton, 2019-03-27:
To include multi-line blocks of code in GitHub use triple backticks, e.g.
````
```
    @property
    def exist_dome(self):
        return 'dome' not in locals()
```
````

renders as

```
    @property
    def exist_dome(self):
        return 'dome' not in locals()
```

AnthonyHorton, 2019-03-27:
@ImmortalRabbit Great to see you adding unit tests as you go! I feel like you're approaching the "Observatory Dependency Decoupling" from the wrong direction, though. Being able to delete the `dome`, `mount` or `scheduler` attributes from the `Observatory` object is not really useful, at least not until the other aspects are done.  Also to do the removal I would simply set the attribute to `None` rather than use `del`, which then makes it easy to check whether it is present (`observatory.dome is None`).

I'd suggest approaching the suggested milestone in order, i.e. start with 
1. **Scheduler Injection**. A first step would be to remove the scheduler creation from the main Observatory class. This will get a user familiar with the overall system and should be straight-forward.

To do this you'd move the creation of the `scheduler` from `Observatory._create_scheduler()`  to a function called something like `create_scheduler_from_config()`, in `pocs/scheduler/__init__.py`, and change `Observatory.__init__()` so that it takes a `Scheduler` object as an argument instead of calling `self._create_scheduler()`.  See `create_cameras_from_config()` in `pocs/camera/__init__.py` for a (much more complicated) example of how this works.

ImmortalRabbit, 2019-03-27:
@AnthonyHorton Thank you very much, I have understood it in a wrong.

ImmortalRabbit, 2019-03-27:
@AnthonyHorton Also, what is about suggested millstone 2: 
Hardware Injection. Remove the remaining hardware components from the Observatory class. Cameras have already been removed so this would involve mostly the mount and dome as well as a clear system for any future hardware.

As I know camera removes using method:
```
    def remove_camera(self, cam_name):
        """Remove cam_name from list of attached cameras.

        Note:
            If you remove and then add a camera you will change the index order
            of the camera. If you prefer to keep the same order then use `add_camera`
            with the same name as an existing camera to to update the list and preserve
            the order.

        Args:
            cam_name (str): Name of camera to remove.
        """
        self.logger.debug('Removing {}'.format(cam_name))
        del self.cameras[cam_name]
```
And here it uses `del` to remove, therefore I was trying to use `del` to remove mount and dome in a same way as cameras were removed. I think I have misunderstood this part too, could you clarify this part too?

Also, I have understood this suggested milestone as to implement test cases for changes that I have done in part 1 and 2. Is it right or is there anything else to do is required?
 
Dynamic injection/removal. In order to test a real system that is running live, help develop a testing suite that will randomly cause failures in some of the various hardware components along with successful recovery or graceful shutdown.

Thank you for any response.

AnthonyHorton, 2019-03-27:
@wtgee & @jamessynge are the main supervisors for this project so would be able to explain this better than me. However, my understanding of 

2. **Hardware Injection**. Remove the remaining hardware components from the Observatory class. Cameras have already been removed so this would involve mostly the mount and dome as well as a clear system for any future hardware.

is that it is _not_ talking about adding methods to remove the attributes. Instead it is talking about removing the _creation_ of the hardware components from the `Observatory` class and putting it in external functions instead. In other words it is doing what I described for the scheduler, but for the mount and dome as well. The whole idea of this project is to move the creation of all of the hardware components out of the `Observatory` class and use Dependency Injection instead, for the sorts of reasons described in [this article](https://medium.freecodecamp.org/a-quick-intro-to-dependency-injection-what-it-is-and-when-to-use-it-7578c84fa88f).

The cameras are a special case. Because there are in general several of them they are stored in a Python dictionary. In that case deleting individual cameras still leaves the `Observatory.cameras` attribute present, as a (possibly) empty dictionary. For the other hardware components I think setting the attribute to `None` when the component is not present is the correct approach.

As for

3. **Dynamic injection/removal**. In order to test a real system that is running live, help develop a testing suite that will randomly cause failures in some of the various hardware components along with successful recovery or graceful shutdown.

the dynamic injection/removal simply refers to added the ability to add or remove hardware components while the system is running, not just when the `Observatory` object is created. The complexity here isn't so much the adding or removing of the attributes, but in getting the `Observatory` to respond sensibly when that happens. I'll leave the others to elaborate on this last milestone, but I will say that you should tackle 1 and 2 first, which will include their own tests.

ImmortalRabbit, 2019-03-27:
@AnthonyHorton Thank you for clear answer and for the time that you have spent to explain to me. 
I will wait for response of mentors of this project idea, maybe they can add something more that can help me to better understand whole idea of this project.
@wtgee @jamessynge

wtgee, 2019-03-27:
Thanks @AnthonyHorton for the answers!

@ImmortalRabbit just to reiterate a point Anthony made:

> he complexity here isn't so much the adding or removing of the attributes, but in getting the Observatory to respond sensibly when that happens.

This is the real challenge here. We do already have an `Observatory.add_camera` but are missing the `remove_camera` and similar methods for the other hardware, which is perhaps the confusing part.

Essentially the `Observatory` needs to run independently in a deamon like mode (this is already done) but should be capable of responding to requests for new hardware and should be able to handle itself properly when existing hardware disappears (because of some problem).

The request for the new hardware would look something like a message (via grpc or zeromq or something - discussion in forum) where the new hardware basically is advertising that it is available and ready to work. The `Observatory` would receive that message and add the hardware appropriately depending on the type of hardware.

Each hardware or item would then have some kind of heartbeat communication with the `Observatory` to let it know that it is still alive. This way when the `Observatory` goes to do, for example, `take_observation`, it will call the command on all the cameras it has attached. If the camera is dead and unresponsive the `Observatory` should have already removed it from the list so that it doesn't try to use it to take a picture.

But yes, a good first step would be the scheduler as that would handle some of the basics of the dependency injection without the complexities (i.e. the scheduler isn't a piece of hardware that can fail).

ImmortalRabbit, 2019-03-31:
I have a question about creation of dome, in a description of task it was said that: 

> Cameras have already been removed so this would involve mostly the mount and dome as well as a clear system for any future hardware

Cameras have a function in `camera/__init__.py`:
```def create_cameras_from_config(config=None, logger=None, **kwargs):
    """Create camera object(s) based on the config.
```
It means that it was already have been removed, and we have to do same thing to mount and dome. But, dome already has a function in `dome/__init__.py`:
```
def create_dome_from_config(config, logger=None):
    """If there is a dome specified in the config, create a driver for it.

```
Does it means that function of dome does creation of it in a same way as camera does?

@wtgee @jamessynge

wtgee, 2019-03-31:
Hi @ImmortalRabbit 

> Does it means that function of dome does creation of it in a same way as camera does?

No. While both cameras and the dome have a `create_xxx_from_config`  function, the difference lies in where we call those functions from. 

The cameras are created *outside* the `Observatory` object and passed as part of the constructor arguments, i.e. when used from the `pocs_shell`:

https://github.com/panoptes/POCS/blob/06fb91daff331296f4c212daa453e4a850b3f900/bin/pocs_shell#L164-L166

The dome, however, is currently created *inside* the `Observatory`:, 

https://github.com/panoptes/POCS/blob/06fb91daff331296f4c212daa453e4a850b3f900/pocs/observatory.py#L59

This is what we mean when we say the cameras have been "removed" from the `Observatory`. That is, the creation of the cameras (a dependency) is no longer tightly coupled to the creation of the `Observatory`. 

So we create the dependency outside the object (`Observatory`) and then "inject" it into the object upon creation. Note that we don't want to necessarily require the dependency injection at the time we create the Observatory, hence the `add_camera` and `remove_camera` methods, which can feasibly be called at any time. The caveat is that the `Observatory` needs to be smart enough to know when it doesn't have *any* cameras.

So now we want to do the same thing for dome, scheduler, mount, etc. You can see the difference in the `Observatory.__init__` method, where the cameras are "added" but everything else is currently "created"

https://github.com/panoptes/POCS/blob/06fb91daff331296f4c212daa453e4a850b3f900/pocs/observatory.py#L45-L64

ImmortalRabbit, 2019-04-01:
@wtgee 
Is it all about only mount, dome and scheduler? Because I cannot find Filter Wheel and Focuser in Observatory class.

AnthonyHorton, 2019-04-01:
@ImmortalRabbit The `FilterWheel` and `Focuser` are optional subcomponents of the `Camera`, so the `Observatory` doesn't communicate with them directly. The may be a case for doing dependency injection there too, but we're concentrating on the `Observatory` class for now.

wtgee, 2019-04-01:
It might also be tempting to remove them all in one big PR but please don't do that. :) With our [In Praise of Small PRs](https://github.com/panoptes/POCS/wiki/In-Praise-of-Small-Pull-Requests-(PRs)) in mind, it is a lot easier (for everyone) to just concentrate on one at a time.

ImmortalRabbit, 2019-04-01:
> It might also be tempting to remove them all in one big PR but please don't do that. :) With our [In Praise of Small PRs](https://github.com/panoptes/POCS/wiki/In-Praise-of-Small-Pull-Requests-(PRs)) in mind, it is a lot easier (for everyone) to just concentrate on one at a time.

Yeah, sure I will not create a big PR